sharpie7/circuitjs1 #540: Bug - Tri-state buffer doesn't work

Komit-cyber, 2021-01-05:
When I disable the  tri-state buffer from working(i.e. putting it in high impedence state) the voltage still passes through as you can see in the following example:

https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgosgCgw8qFCrcAWEDNEdqKaCGHjD4tADIhCbVtymtshPlQBmAQwA2AZwCm1OhOatIHBGw68qKjTr20Asp268ciiyBTQEtIA

I would be glad if you could fix it! I need Tri-states for a project i work through this time, thanks for reading. 
Have a nice day everyone.

pfalstad, 2021-01-05:
Looks like it's working ok.  It's in high impedance state, not infinite impedance.  There's nothing pulling the output down to ground, so I see no reason why it shouldn't be high.

Komit-cyber, 2021-01-05:
I understand what you say, its just that I want to isolate a register from the bus and the tri-state buffers don't work the way I thought. For example take a look at the circuit in the photo:

[image]

I want to isolate the register so that it does not drive any other circuit in the bus (for example this OR gate). The tri-state buffers are not enabled and still the output drives the OR gate.
Do you know any way to do this? I would be glad if you could help. Thanks!

pfalstad, 2021-01-06:
The OR gate in the simulator has infinite input impedance, so it doesn't draw any current at all..  But the tri-state does not have infinite output impedance.

If this were a real circuit, and the tri-state buffers were not enabled, the output would be floating, since nothing else determines its state.  So if you really need the OR gate output to be low in this case, it doesn't seem like a good idea to have the bus lines floating.  Use an AND gate instead of a tri-state, but then no one else can drive the bus, of course.  Or use a pulldown resistor like so: https://tinyurl.com/y2eyar59

Komit-cyber, 2021-01-06:
Ok, got this. My bad.Thanks a lot for your help!

pfalstad, 2021-11-19:
I should probably fix this.  This is a common complaint.

closed 2022-01-15 by pfalstad