sharpie7/circuitjs1 #545: change capacities

maulem222, 2021-01-15:
In the offline editor version, how do I add a real-time ruler for the capacity of resistances/capacitors/inductors ?

[image]

pfalstad, 2021-01-15:
Right click on an element, select "Sliders..."

maulem222, 2021-01-16:
A current of 50V flows through a capacitor en charges the capacitor. When the switch is opened, the current is interrupted and the capacitor starts to decharge. When I add an output to the capacitor circuit, it shows negative and positive voltages almost double/triple as high as the source. How is this possible?
[image]

sharpie7, 2021-01-16:
The inductor will generate back EMF when the switch is opened and the current through it is removed. It is the principle of many circuits that need to raise the voltage like Boost convertors and CRT High Voltage supplies.

maulem222, 2021-01-16:
Yeah thanks. Soon after posting this I realized that it must be the inductor generating the missing voltage. Simply forgot that is was there.

Another question. Is there a simple way to stabilize the resulting alternating current flow so that it doesn't stand still after a while? Or should I use alternating current instead of decent, in order to get a stable oscillation? Can alternating current be transformed into other wavelengths by the use of capacitors and inductors?

sharpie7, 2021-01-16:
> Is there a simple way to stabilize the resulting alternating current flow so that it doesn't stand still after a while?  Can alternating current be transformed into other wavelengths by the use of capacitors and inductors?

Neither of these things can be done with passive, linear components (resistors, capacitors, inductors). There are ways to do both of them with active components (like transistors). These are areas where you probably need proper training or reference materials if you are going to understand them throughly. The examples in Transistors->Osciallators will give you a flavour.

closed 2021-01-16 by sharpie7

maulem222, 2021-01-17:
Today, I studied pnp and npn transistors by use of reference materials online. I found a schematic of a blinking led and drew it in circuitjs.

When the circuit is running, after a while the led starts to blink on and off.

With regard to this circuit, I have the following explanation for what is happening. 

First, the current through the resistor charges the capacitor, until the current through the capacitor stops. Then the capacitor is decharged as a matter of current flowing through it in the opposite direction. This causes current flowing from the capacitor to the pnp transistor, which starts to conduct, and in turn activates the current through the second (npn) transistor. This causes the larger current to flow for a short while and stop again, because the pnp transistor doesn't conduct as a result of shortage of current at the emitter. This in turn is effected by the capacitor again becoming saturated.

Correct me if I am wrong.

[image]
[led blinking.zip](https://github.com/sharpie7/circuitjs1/files/5826329/led.blinking.zip)

pfalstad, 2021-01-17:
The current through the capacitor is oscillating; it switches direction at every step.  This is not what would happen in a real circuit.  To fix this, make the time step smaller (using Options->Other Options), increase the size of the resistor, or edit the capacitor and uncheck the "Trapezoidal Approximation" box.

Where'd you find the circuit?  Sometimes circuits you find on the internet need some tweaking because they depend on non-ideal behavior of the components.

You might want to use File->Export As Link when posting circuits; it's easier for everyone.