sharpie7/circuitjs1 #547: Feature-Add common ground symbol

gosh-darn, 2021-01-17:
It’s great to see the new signal and chassis ground symbols, but another symbol that is commonly used is what I usually refer to as the common ground symbol.

It looks like the regular earth ground symbol without the two smaller lines. Here from a 1954 radio schematic:

[image] [image] [image]

pfalstad, 2021-01-17:
This is already implemented, but only sideways 🙂

gosh-darn, 2021-01-17:
Well that’s a very good start 😃

closed 2021-02-21 by sharpie7