sharpie7/circuitjs1 #548: The addition of NTC resistors

Bytesforus, 2021-01-17:
They are currently not present in the simulator it seems.

pfalstad, 2021-01-17:
They're there, under Active Components->Thermistor.  I guess they should be passive components.

Bytesforus, 2021-01-18:
Good stuff, thanks. The standalone version I installed a few weeks ago does not, missing also the photoresistor.

A minor comment regarding the beta value for MOSFETs, to my knowledge this is a characteristic for transistors only.

closed 2021-01-18 by Bytesforus

pfalstad, 2021-01-18:
It's a parameter used to model a MOSFET as well.  See here halfway down, "basic DC equations":

https://ece-research.unm.edu/jimp/vlsi/slides/chap2_1.html

Bytesforus, 2021-01-19:
> 
> 
> It's a parameter used to model a MOSFET as well. See here halfway down, "basic DC equations":
> 
> https://ece-research.unm.edu/jimp/vlsi/slides/chap2_1.html

Some files in the standalone version seem to have paths that are too long in windows.