sharpie7/circuitjs1 #549: Support for SPICE models?

Harvie, 2021-01-19:
I know this might sound as too complex project. But i wonder how hard it would be to implement "generic SPICE" component that would be described by SPICE model rather than java code. I am not talking about using SPICE backend for whole simulation. But just being able to do calculation of that single component in single time step based on user-specified SPICE model.

I wonder if there is some kind of "spice model interpreter/solver" written in java which would be easy to integrate as new component class in the circuitjs. I beleive it is certainly worth the research to look for such java SPICE implementation.

Circuitjs has UI/UX that is waaay better than anything i saw in SPICE simulators. But sometimes i lack possibility to use some specific components. This might help. Also it would open the doors to use of thousands ready made models including various digital ICs.

pfalstad, 2021-06-22:
I've been doing some work in this area but it's a big job.  In addition, I've found it surprisingly difficult to find SPICE models to test.  Maybe I just don't know where to look.

If you some particular devices you need, post them here and I will see what I can do.

Harvie, 2021-06-22:
Rather large and useful collection of SPICE models can be found here:
https://github.com/kicad-spice-library/KiCad-Spice-Library/tree/master/Models

I don't really know how SPICE works, but to me it seems there are two kinds of models. One that mathematicaly describes behaviour of the component using some equations. eg.:
https://github.com/kicad-spice-library/KiCad-Spice-Library/blob/master/Models/Optocoupler/PS2561F.lib

And one that just reuses some primitive components built in SPICE. For example most models of BJT transistors are simply defined as parameters for generic BJT model present in simulator, eg.:
https://github.com/kicad-spice-library/KiCad-Spice-Library/blob/master/Models/Transistor/BJT/BC546.lib