sharpie7/circuitjs1 #550: LDR Photoresistor made sensible for a LED light

ThorleifBu, 2021-01-19:
I would like to suggest a modification of the LDR/Photoresistor component.
Today it is controlled by a slider.
I suggest an option such that I can enter a LED color value, ie. "RGB Sensitive [100]"   (Default value = 000 = Use Slider)
This would then disable the slider, and instead, the LDR should get its information from "That LED" in the circuit where the same color is selected.
(_If more than one LED with that same color, then just pick the first in the circuit code_)

Basically this will enable a construction having several LED in a normal way. But having a small separate circuit (ie. an oscillator with a LED) I could then set that LED to an unused color, maybe Yellow = 110.
Entering the Yellow code into the LDR, the LDR would then respond to the light intensity from the LED.
Sensitivity could be set so that No light gave max resistance of the LDR, while max LED light gave the minimum LDR resistance.

An additional suggestion might be that the edit of the LDR component could also have two value boxes for [Min resistance at full light] and [Max resistance at dark].

pfalstad, 2021-01-19:
That sounds like you want to make your own optocoupler.  We have one already, under Active Building Blocks...

ThorleifBu, 2021-01-19:
Well, strictly speaking yes. But the issue came when a group of students wanted to simulate a visible light data transmission circuit based on the LDR resistor, and not the photo transistor of the optocoupler.
Im many small light sensitive circuits the LDR's resistor characteristics, makes it a nice component.