sharpie7/circuitjs1 #551: Feature request: Some common RF components

Quadrapod, 2021-01-20:
Some fairly basic RF circuits and principles can be difficult to demonstrate before the simulation starts to really slow down. Multiplying mixers are one of the worst offenders where a single Gilbert cell or even diode mixer can be enough to slow things down pretty substantially.  This is particularly troublesome since in many cases the IF and LO/RF can differ by orders of magnitude meaning after implementing some basic filtering it can take a lot of waiting with the applet jittering along to get a resultant waveform to show.    

A few simple ideal RF components would go pretty far toward making certain principles easier to demonstrate without taxing the simulation as well as allowing the creation of signals that are difficult to generate at the moment using existing options.

pfalstad, 2021-01-20:
Which ones in particular?  A multiplier can be made with a VCVS like so: https://tinyurl.com/y2yjkvl6

Quadrapod, 2021-01-21:
I wasn't actually aware of the VCVS or how it functioned.  That covers pretty much everything I was having issues with.  Thank you.

closed 2021-01-21 by Quadrapod