sharpie7/circuitjs1 #552: Feature Request: intermediate switch ("Kreuzschalter")

MisterHollandt, 2021-01-26:
since corona I'm using the circuit simulator applet in my electro-technology online-classes.
It's a great way, to actually show my students the flow of electricity in our simple electrical installations.
And they can try out their own circiuts from really any device.

Therefore I would greatly apprechiate an **intermediate switch** for the cross connection (I hope this is the correct term).
(The intermediate switch would be the same as two changeover switches ("Wechselschalter"), that are switched synchronously.)

Thanks in advance

pfalstad, 2021-01-26:
You mean one of these?  http://www.falstad.com/circuit/e-4way.html

MisterHollandt, 2021-01-27:
Yes, exactly - thank you for the quick answer!

 

Is this synchronized "double-two-way-switch" in one of the submenus? I just can't find it.

Since I want my students do construct and simulate their own circiuts, they would need this switch

 

(Ofcourse - as a workaround now I can give them the reduced circiut as export/import-txt-file)

Thanks onxe more

 

---
Im Röhrgewann 2
64367 Mühltal
0179 - 6681468

 
 

Gesendet: Dienstag, 26. Januar 2021 um 20:46 Uhr

pfalstad, 2021-01-27:
Hi, you just create two switches and then synchronize them by right clicking, Edit, and set Switch Group to 1.

closed 2021-01-27 by pfalstad

MisterHollandt, 2021-01-27:
Okay, that works fine - thank you!

 
 

Gesendet: Mittwoch, 27. Januar 2021 um 17:50 Uhr