sharpie7/circuitjs1 #553: Feature:Add Power Electronics Exsimple

annopnod, 2021-02-02:


annopnod, 2021-02-02:
buck converter model
$ 1 0.0000010000000000000002 2.1831051418620846 50 5 50
r 256 -48 256 112 0 2
r -144 -16 -224 -16 0 10000
r -144 -16 -144 64 0 100
w -160 -48 -224 -48 0
w -224 -48 -224 -16 0
R -352 -48 -432 -48 0 0 40 12 0 0 0.5
a -224 224 -352 224 8 12 0 1000000 9.306390106060656 8.52 100000
a -416 240 -544 240 8 12 0 1000000 -0.00007863901060606545 0.14400001880077085 100000
t -352 32 -304 32 0 1 -11.962053119327225 -11.962053113927455 100
t -352 80 -304 80 0 1 -11.962053113927455 0.03794688067277592 100
t -400 96 -352 96 0 1 0.5977734955161277 0.6357203761889036 100
w -352 32 -352 80 0
r -352 32 -352 -48 0 4700
r -400 96 -464 96 0 10000
w -304 16 -304 -48 0
w -352 -48 -304 -48 0
w -304 -48 -224 -48 0
w -304 48 -304 64 0
w -352 112 -352 128 0
w -352 128 -304 128 0
w -304 96 -304 128 0
g -304 128 -272 128 0 0
w -304 64 -144 64 0
r -224 208 -128 208 0 100
c -352 160 -224 160 0 1.0000000000000001e-7 -9.306468745071262 0.001
w -352 160 -352 224 0
w -224 160 -224 208 0
w -416 224 -352 224 0
w -464 96 -544 96 0
w -544 96 -544 240 0
R -416 256 -352 256 0 4 80000 6 6 0 0.5
d -80 112 -80 -48 2 default
w -128 -48 -80 -48 0
l -80 -48 -16 -48 0 0.00009999999999999999 4.9234302190726e-9
g -80 112 -80 160 0 0
w -16 -48 32 -48 1
w -128 208 128 208 0
172 -224 240 -224 320 0 7 8.52 12 0 0 0.5 Voltage
w 128 208 128 -48 2
w 32 -48 128 -48 0
f -144 -16 -144 -48 41 5 100000
w 128 -48 256 -48 0
g 256 112 256 160 0 0
c 32 -48 32 112 0 0.001 9.259858181425571 0.001
w -80 112 32 112 0
x -205 251 -66 254 4 24 cut_off\sfsw/5

[image]

annopnod, 2021-02-02:
spwm
$ 1 0.000001 58.6985430941709 50 5 50
a 544 160 672 160 8 15 0 1000000 3.179999732768546 -0.4651169397850349 100000
a 544 80 672 80 8 15 0 1000000 3.179999732768546 0.4651169397849704 100000
w 544 64 464 64 0
w 464 64 464 144 0
w 464 144 544 144 0
R 544 96 368 96 0 1 50 6 0 0 0.5
R 544 176 368 176 0 1 50 6 0 3.141592653589793 0.5
R 464 144 464 272 1 4 10000 3 3 0 0.5
O 672 80 768 80 0 0
O 672 160 768 160 0 0
o 8 64 0 12290 15.000593985004837 0.0001 0 1
o 9 64 0 12290 15.000593985006562 0.0001 0 1
[image]

pfalstad, 2021-02-02:
Well, we already have buck and boost converters under "Power Converters"...

closed 2021-02-16 by sharpie7