sharpie7/circuitjs1 #554: Feature Request: More electrical elements

MarinoHappiness, 2021-02-04:
Hello,

Thank you for adding power converters and DC motor in the app. Please keep it up. I often use your app to learn about electronics and I hope you will add other good electric elements and circuits, such as resonant converters, BLDC motor, IGBT, Arduino, sensors, etc. 

I had no success with simulating resonant converters. Can someone give a link to a good resonant converted circuit?

Will you be creating elements of mathematical algebra for the app (for feedback/control loops)?

Thank you,
Marino

pfalstad, 2021-02-05:
What do you mean, elements of mathematical algebra?

MarinoHappiness, 2021-02-05:
[image]

The [mathematical models](https://en.wikipedia.org/wiki/State-space_representation) used for designing feedback/control loops in electrical [control] engineering.

pfalstad, 2021-02-06:
Well you can use a VCVS to do an adder: https://tinyurl.com/y54mfw2k

And there is an integrator example under Circuits->Op-Amps->Integrator.

MarinoHappiness, 2021-02-06:
I see. Thank you for creating this element. I hope you will expand the Controlled Source Output Function with other mathematical functions (integration, constants, differentiation, etc.) for simulating PID control.

pfalstad, 2021-02-06:
You can put constants in there, or any other mathematical function except for integration and differentiation.

MarinoHappiness, 2021-02-06:
Why excluding those two? Doing integration and differentiation in PID control with operational amplifiers turns out to be an over endeavor. I think it's easier to define the mathematical functions of PID control using only Controlled Source Output Function elements. See what I mean: [https://tinyurl.com/y4rbykjz](https://tinyurl.com/y4rbykjz)

pfalstad, 2021-02-06:
If I added integrators/differentiators, I would probably do it as separate components (not part of the VCVS) because they require state.  They're not just simple functions of the input.  For now you can create a subcircuit and encapsulate the op-amp implementation to make it smaller and more convenient.

IPlayZed, 2021-02-11:
I was wondering if there is a way to create multi-emitter BJTs, or if I just haven't found it?
Thanks!

pfalstad, 2021-02-11:
No multi-emitter BJT's, sorry.

pfalstad, 2021-02-24:
You use a VCVS to do integration/differentiation now.  http://www.falstad.com/circuit/customfunction.html