sharpie7/circuitjs1 #557: Feature Request: Relating Electromechanical to Electrical

KA300SE, 2021-02-19:
Would it be possible to allow certain electromechanical components to affect electrical components by way of changing their values? For example, what if the motor was able to increase in temperature, and that increasewas reflected through a thermistor. It would be as if the thermistor was in the vicinity of the motor windings, or perhaps the motor was representing a heat blower motor. Then, both components could be present in the circuit but also contribute to a larger control system like a PID.

TimFFischer, 2021-02-22:
I just wanted to open up a kind of similar feature request, but will support this request with a similar problem field: Is it possible to implement additional values like energy flow (i.e. power) as an output value?

The reason for me: When calculating the power with a voltage controlled source (or a wattmeter, see #561), the output will be written as Amps or Volts, which is kind of irritating. It would be better to have the correct physical representation, or at least no incorrect one. Physically, the best thing would be for power a flowing particle like for the charge flow (but as a physicist, I would raise the question to myself which potential representation should be used? Temperature T^1 or T^4? ;-) ). As a developer I also see the "endlessness" of this idea: it becomes more complex to parameterize the interactions, connect all physical in/outputs and check at which physical genre to end: thermydynamics? Hydraulics (p, V) ? Photonics? Mechanics (M, phi) ? Shall the momentum be a represented as a flow?

As an alternative solution to this problem: Is it possible to implement a non-voltage / non-current value representation (mabe with freely choosable physical unit)? This might also be useful in digital circuits to input / output absolute values (so from %00100101 to 37, where 37 has no value). With some new component for type-casting unitless to voltage, current, bit and vice versa, one could piece together own solutions with custom subcircuits.

pfalstad, 2021-02-26:
Ok I understood what KA300SE said but I don't think I'm likely to implement it, because I'm not sure how to come up with a model for the temperature of a motor, and if I did, I don't think that would be useful for most people.  We could potentially give you access to the speed of the motor somehow, and then you could come up with your own model using a VCCS.

Tim, I don't understand what you're asking for.  Except for the part where you want the wattmeter output to be displayed in watts.  That seems reasonable.

KA300SE, 2021-02-26:
I guess essentially what I was envisioning was a way to connect output values of the motor directly to another electrical component somehow. Having the user make their own calculation model is perfectly understandable. I was just hoping that I could have a motor attached to a PID controller that I made, and the motor could affect say a thermistor. Instead of me moving all the slider to simulate a change in temperature or speed, the motor's output values could do that and I would have a completely self-sustained system.