sharpie7/circuitjs1 #559: Feature Request: Scope with trigger

roy77, 2021-02-21:
The scope support x/y an rolling view. But the common variant of a triggered view is not supported. A simple edge based triggered view of offset adjust would be nice.

sharpie7, 2021-02-21:
Agreed. It's on my to-do list!

Jeppe53, 2021-08-07:
Could be fine if the time is set to zero at the tigger point.

pfalstad, 2021-08-07:
For now, you can attach a stop trigger to something (under Outputs and Labels) and let the simulation run until the condition is reached.  You can also set a delay so that the simulation runs for a short while and then stops after the trigger event, or you can just slow the simulation down at that point, remove the trigger and let it run for as long as you want.

vcolella, 2024-09-12:
Is anyone working on a PR for this ?