sharpie7/circuitjs1 #560: Bug- uncorrect behaviour of the ideal op amp

TimFFischer, 2021-02-22:
I'm struggling with the open loop transfer charateristic of the ideal operational amplifier with a single supply (= unipolar).

Currently, the output voltage is 0V for differential input voltages smaller than 0V. As far as I understand the output stage of op amps that is not the case. The output voltage for a differential input of 0V should be 0.5 * (Vs+  -  Vs-). With a positive supply voltage Vs+ of e.g. 15V and the "negative" one being 0V, this leads to 7.5V for a differential input voltage of 0V.

The situation of Vs+ = 15V and Vs- = 0V with a differential input of 0V on the ideal op amp should be similar to the situation, when all voltages are shifted with -7.5V. This would lead to Vs+=7.5V, Vs-=-7.5V with a differential input of 0V but the output voltage shifted with -7.5V. Since the output voltage of the dual supply op amp is known to be 0V for differential input of 0V, the output voltage of the single supply must be shifted with +7.5V.

Maye the [simulation](http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l5YCcyWrQDitAzAFkZHgEyICsA7LgGxjq5hUimRMi7ZMCmAtGGAFABDENyq4Q6FtiISWicKSjh48JTHhhm6ZkXI18pRESoywKyPwDuIabLaZJUKyLHgimUeMSMLAJRfirrxgMq4sELgs6nBIaHEomFFYpPwA5gFuHq5guOJR-JEcIZi22NiMDEnYAPpi1ZDVAB7o2JAAJkTVYDzkXTyI9dWd0XDY8eNUg2CDpPX8RCEijpEs3LYrSooAxgA2ggC2AA4AFFsAVMeC3ABGAJTcxzxttwD0RAA0nO-PaeCM7KZ-rhEk50txlrZwSxciC-EtoXARMU2IjwiisBpxvFYclnAwUVIZBs4VC2EFkdQ8kpIsoVGpcf5uN42MzeO4WT5qeFFEkYClGczSOy2ZghTiabzcYVMgSbOUmOy5LVcINmq0Ol1oFRzDr4NgeKQVWBYLrdd1uBRBsMTaMsXFJg0zDM5tYmf9WctmRZXYLhY4xU5GiIiAsyeJeMgyeQ2G4QKlBABLAB2-CD3BD0MQHk0ooQMcWAFlBI0ADoAZwA8gBXAAuh1rqeDRBYpFaSJMTDK+ZkBeT5erdYbYMc0lWI-l+UZlRAIfDzcws6UESiNtidoS9L5zlJ7g8jiIpB8eMYRBptlP+WEO5p+IvIHk5Ggim4j8UZnMIFfn6fMS0Oj0mi0vA-AAG7tiwi47qe9I0l+vJPqB8I2M2SG7jBLBwWoCFgeeNKuHeSSwU+m4IdY+E0vuEoFIQMrznKjBlJyCzKqq6BmBqswwNMDTWioYzrigDpDOgzoWGmYziGYwbUOAHCSTIhwAE6cGWnBJjW5bHMmWwAPaKcpWw1gA-Lc-aHAAgkcjZrIgEDYOQMjpoaNjkIo8kgLp+mcIZZmWYcmkgQmgjlvsgg1gAFrc1ljJgDAeM2EBgOQESxp5BkaZWFlHAFQXlocOk1mpNZBTs5ZluFCYAGbFUmqRRWmZ5FAx-yxgAamWJb+R1ACO5aaK10XyCED6yUSbVltw5Ylr1ZaQANDULnmfAzpA4jue1nVTTNr7zSIuCKO40aASQ0LjZNPXli+T4DTpM6YOIcmQFmK1YBA4T8EAA) explains the situation a bit better.

Could you have a look for it and correct the ideal op amp?

pfalstad, 2021-02-22:
Hmm, that's not good.

ormaaj, 2021-04-02:
Define "ideal". Your opamps have floating outputs. Ideally when the inputs are equal, the impedance to both supply pins is infinite and the output is "off". With a floating output, voltage is only determined by subthreshold effects and/or the simulator's "gmin" and "rshunt", or just produces an error.

Bipolar opamps are current in and current out. Common macromodel output stages usually utilize controlled current sources as output elements, though other implementations are possible.

They do commonly have an internal reference which may be defined as `(VCC + VDD) / 2`, but its purpose has nothing to do with the output. As far as I can tell it's usually there to define a reference for parts of the model that are otherwise isolated from the inputs, rather than using an input or the global ground node directly. That can make it easier to verify correctness and help prevent (or at least debug) side-effects that might creep in if inputs or node 0 were used as a reference.

Simulators may also do some tricks to reduce float error by keeping voltage / current values shifted closer to zero in the case of a ground-referenced source. That would be an implementation detail.

pfalstad, 2021-04-02:
I think ideal means like the VCVS version, where the output is a voltage source proportional to the difference in the inputs, plus some base value.

ormaaj, 2021-04-02:
So it's just a question of what base value makes sense? That value would really only come into play when the output is connected exclusively to high-z nodes for initial conditions calculation though right? Shifting the output to the supply midpoint as the input difference crosses through zero would make the opamp behave like a digital "magnitude comparator". I think if it's open loop the zero-value output should be determined by the last known state, and the only exceptional case should be initial operating point analysis.

pfalstad, 2021-04-02:
Yes, that's probably why nobody noticed until now.  I'm not sure what the application was.  But using the midpoint of the supplies seems more reasonable than using ground.  Doing what you describe would be much more difficult for a very exceptional case.

ormaaj, 2021-04-03:
It's not so noticeable with the fixed supply and symmetric zero-centered default. Or it would implicate anything with zero capacitance high-z pins, but the behaviour of those nodes really affects all parts.

It looks like you've already done something about that for e.g. [MOSFET C*O](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgpABZsKBTAWjDACgA3EQmkbNbrzR4oomlSRUp0BGzAJi4FCOGCQqqgLDwd8NgDMlK-OEKaT-cDKiwUbAO6mq-Kjz4DIbAOZqXv7ISingBKIMSBYMQoYRFm4OrImtYo1tKyYITRGZoovNnumiAAJgz6AIYArgA2AC5MVQxF8VI2kOyObqpuYMpQDmo9IvmDfQBOTkYTkdFU2nBsQA)... a different issue there.

pfalstad, 2021-04-03:
You're talking about the MOSFET gate voltage?  The analog switch is never a broken circuit like the UI suggests.  In it's off state it's just a 10Gohm resistor, so that keeps the gate voltage from being undefined.

Allowing the op-amp to have a high-impedance output state would require me to add an extra internal node (so we can add a variable resistor to the output) which I'd rather not do.

pfalstad, 2021-04-24:
fixed

closed 2021-04-24 by pfalstad