sharpie7/circuitjs1 #561: Feature- new component: wattmeter

TimFFischer, 2021-02-22:
Is it possible to implement the wattmeter as new component? (with 4 terminals, as seen in the [schematic symbol](https://images.app.goo.gl/q2T9B2GHwS4qiLHf7))

In order to analyse the utilisation factor and efficiency in my online script for the first semester, I calculate the power from the current and voltage ([ugly googly-english translation](https://translate.google.com/translate?sl=de&tl=en&u=https://wiki.mexle.org/elektrotechnik_1/lineare_quellen_und_zweipole%23leistung_an_zweipolen_und_bezugsgroessen), translated [simulation](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0mQrFaB2AzADgJwIGyU1qpDvBPOiIgCyqUCmAtGGAFABuIVFATFZJxTBUqUUX0qj+MeCwDuA8MJCYcikZBYAnZaqEiweNaLBwN87pEFLe-PVC0hUF8MlVP+N43DmPn3ZNwgFjwB9vIGHqFcQaFmCv6B0Z4a2tzoViKedlKwkKzcYIlOIDgiVKUl6o4gABQAhgwARgCUAPQwcGAAVDWNDADGzT6eFTSBFXHuRhj6SpN+oTMxgXFjQbycxYUUGgDmlGk6B4KFkiwWyMeVlIajIAAKAPoAMsPizIFZc2-8FVmnk3QIhUjiBRlWxW4GzWCXsjXAulOYGQmCCBDhQUsvkomAgBnULAAHutMvwMDxaPoKAAHAD2sjomgAOgBnAC2dDqLIArpo6ByAHYAF2GWLsEXBopCgQlsI01NBsxES1hECkPimdk1rnsqGQZMgSt8tiUarCxqC4imco1YJBaxBgMyUUhsSJCJA6AghjBVJAABsAJYCzmad04VQWVTIHj4TjgCgAVXd6E+p3SQTA1HAK3dzH0oUKumQ2bAFAASnmiC5BFsvfHToH3WlI8w0TwhA2K+6iB4DI5iJnUfpAq9idx4KjmJHkHiwMOEyBK+O0DmPPXCpScyAAMrDDNaw2S4moSeewJ6iDoUtJnuzz1uFGehenPfEviXecUKhgWgWCAjiAY6OJgtDzpSCDgPai7JieJg5he-aFKWo49pAYH9kQJwvhQACSeaqJgU7wIgJiJIubC0v6Qp1LsdCsiytK8v0dAsLS4CkcYULYEgcBRjgzDzuUwQYIgHSdKIMoEux+hiAYYHQBOjCCIp8CMPBEDkagLAYPGthwJwphAbSdQACasuWLwsEAA) ). In an other simulation, I try to show the importance of the series resistor when charging a capacitor (googly-translated "Exercise 7.2.5 Exercise on the charge equalization of two capacitors" at the end of [this site](https://translate.google.com/translate?sl=de&tl=en&u=https://wiki.mexle.org/elektrotechnik_1/lineare_quellen_und_zweipole%23leistung_an_zweipolen_und_bezugsgroessen)).

pfalstad, 2021-02-22:
Someone asked for this already: #431

Then we thought maybe it would be better if it were possible to plot the sum of scope values, so then you could plot the power consumed of several elements and take the sum.  So that's the direction I was thinking of going.  But perhaps a wattmeter would be easier to use, and certainly simpler to implement.

pfalstad, 2021-07-11:
added

closed 2021-07-11 by pfalstad