sharpie7/circuitjs1 #562: Feature - Starting Voltage for capacitors in example circuits and the new crystal.

evan-Slater, 2021-02-22:
I noticed that in some of the circuits that have capacitors, the capacitors have an initial charge to them, But this is only when the circuit is initially loaded, not when reset is pressed. would it be possible to fix this?

Also, I noticed that the crystal workers allot like the capacitors, and will have an initial charge in the circuits it is in. But there is no way to add said initial charge to it. would there be a way to add that to them.

pfalstad, 2021-02-22:
You can right click a capacitor, select Edit, then set the initial charge to set the capacitor to when reset is pressed.  Unfortunately you can't do that with crystals yet.

pfalstad, 2021-02-26:
Is it that important what state a crystal starts in?  Crystals are used for very high frequency oscillators, and the initial state doesn't seem that important unless I'm missing something.

evan-Slater, 2021-03-01:
I have looked at the premade circuits that use the crystal, all of them seem to have the crystal start with a charge, I have been trying to get them to work when the reset button is hit but this one fact seems to be the only obstacle now. It is not a major problem, I just use this site to try and brute force learn circuits.

pfalstad, 2021-03-01:
I could remove the initial charge on crystals.  That's only there because capacitors have a default 1mV charge on them when reset, because many oscillator circuits won't start up otherwise.

pfalstad, 2021-03-01:
Here is a crystal that has no charge on reset: https://tinyurl.com/ycb4m3ze

You can cut and paste it into your own circuit.