sharpie7/circuitjs1 #564: Comparator does not work properly

Harvie, 2021-03-03:
I am pretty sure the output of comparator should be high in this case:

[image]

Anyway it does weird things all the time.

Code:

```
$ 1 0.000005 10.20027730826997 50 5 50 5e-11
401 256 272 400 272 1 8\s15\s-15\s1000000\s5\s2.5\s100000 0\s20\s10000000000 0\s0
R 256 256 208 256 0 0 40 2.5 0 0 0.5
172 256 288 208 288 0 7 5 5 0 0 0.5 Voltage
401 592 256 720 256 1 8\s15\s-15\s1000000\s4.999848915455328\s0\s100000 0\s20\s10000000000 0\s0
r 464 272 544 272 0 1000
w 544 272 592 272 0
r 720 336 592 336 0 1000
w 592 272 592 336 0
L 448 416 368 416 0 1 false 5 0
w 448 416 592 416 0
d 592 336 592 416 2 default
w 592 240 592 192 0
w 592 192 256 192 0
d 720 256 720 336 2 default
```

Harvie, 2021-03-03:
Ah, might need pullup actualy...

closed 2021-03-03 by Harvie

pfalstad, 2021-03-03:
I completely forgot this element existed.  Yes, it needs a pull-up.