sharpie7/circuitjs1 #565: Feature: arbitrary web link (not just dropbox)

fordsfords, 2021-03-15:
Hi. One (easy?) way to import from any web hosting service (e.g. GitHub) would be to just allow any arbitrary URL that is a text file. (The GitHub "gh-pages" feature can be used to make text files available as directly delivered via user.github.io, rather than decorated by GitHub's UI).

pfalstad, 2021-03-15:
Well there's no UI for it, but you can load any arbitrary file by specifying it in the URL when loading the page:

http://www.falstad.com/circuit/circuitjs.html?startCircuitLink=http://www.falstad.com/circuit/circuitjs50/circuits/ladder.txt

The file must be on a server that supports CORS access, and it seems that GitHub pages does.

fordsfords, 2021-03-15:
Perfect! No need for a UI.

I tried it with one of my pages and it worked:

http://www.falstad.com/circuit/circuitjs.html?startCircuitLink=https://www.geeky-boy.com/test.circuitjs.txt

(That "geeky-boy.com" page is hosted on GitHub.)

Anyway, I'm good with closing this. Thanks!

closed 2021-03-15 by fordsfords