sharpie7/circuitjs1 #566: Incorrect Circuit: Parallel-In Serial-Out

Zhongy1, 2021-03-15:
This is specifically for the circuit in Sequential Logic > Shift Registers > Parallel-In Serial-Out.
Here is what the circuit currently looks like:
[image]
The problem with this is the first input is unaffected by the load/shift input, causing the circuit to fill up with with a logic 1.

I added a mux in the front of the left most D flip-flop which seemed to have fixed the problem.
[image]

pfalstad, 2021-04-25:
fixed

pfalstad, 2021-04-25:
thanks for the fix

closed 2021-04-25 by pfalstad