sharpie7/circuitjs1 #568: Strange capacitor behavior with default trapezoidal approximation

quetzalcoatl, 2021-03-22:
On: https://www.falstad.com/circuit/
About: Circuit Simulator version 2.4.3js

Circuit:
`$ 13 0.000005 0.08464817248906141 50 5 50 5e-11
v 96 336 96 64 0 0 40 12 0 0 0.5
S 256 336 192 336 0 1 false 0 2
w 96 64 256 64 0
w 368 32 400 64 0
w 256 64 192 160 0
c 256 64 256 336 0 4.7000000000000005e-7 11.997908748066372 0
r 96 336 192 320 0 0.1
w -128 -16 32 -16 0
w 336 336 400 336 0
d -112 16 32 16 2 1N4004
d -96 0 32 0 2 1N4004
w 192 240 256 336 0
r 400 336 400 192 0 165.82
l 400 64 400 192 0 2.97 0 0
r 192 160 192 240 0 1000000
c -96 48 -16 48 0 1e-12 0 0
w 368 32 336 64 0
r 96 336 192 352 0 1
o 5 1 0 4103 20 3.2 0 2 5 3
o 13 1 0 4103 0.0000762939453125 0.00009765625 1 2 13 3
38 5 0 0.000009999999999999999 0.00101 Capacitance
h 2 6 5
`

Sorry about the loose parts laying around, I was in the middle of testing something else.

The odd thing happening is, when the switch is set at 1R, the capacitor charging time  is about 50us, and when the switch is flipped to 100m=0.1R, charging that same capacitor takes.. 485us? That's pretty odd. When the resistor is changed to even lower value, like 10m=0.01R, it's even worse, current and time rises up to the sky.

[image]

It seems like the voltage source and the capacitor enter some ping-pong ringing state: capacitor is overcharged by a tiny amount, then discharged too much, and re-charged, and so on. At first I thought the 'wires' have some inductance modelled-in :), but I played around and changing the timestep from default 5us to 0.1us helps a lot (while leaving default 5us and turning on 'auto-adjust timestep' makes almost no difference).

Then, I noticed this happens only when the capacitor has "Trapezoidal approximation" option turned on, which, frankly, is by default on.

Btw. I'm pretty sure there is a similar problem with inductors, and disabling approximation also helped with that.

I understand what approximation is :) and I don't know if this behavior is a bug, or is a sideeffect of the approximation and time-stepping, but if disabling approximation (even with the same time step) makes these problems go away, if it can make such a huge difference even for simple circuit like R-C, should this approx be turned on by default?

pfalstad, 2021-03-22:
It's a side effect of the approximation.  Maybe trapezoidal should be off by default.  I could have sworn I tried the LRC circuit with trapezoidal turned off at some point and got very bad results.  But I just tried it now and it was fine.  I'll have to look into this.

pfalstad, 2021-03-22:
The time constant of this circuit is 47 ns but your timestep is 5 us which is much too long, so it is not possible to simulate the charge of the capacitor accurately.  The trapezoidal approximation tends to overcorrect in cases like these which causes the ping-pong you are seeing.  If you set the timestep to 5 ns it works fine.

quetzalcoatl, 2021-03-22:
Oh god. sorry about that. I miscalculated it as ~45us not ns. Yeah, that makes much sense now.

closed 2021-03-22 by quetzalcoatl