sharpie7/circuitjs1 #569: Suddenly cannot press right mouse button on certain components

ferrybig, 2021-03-22:
I was working on this circuit: https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKABMQ84QAWbjQnhDYEKEGLZ0AZgEMArgBsALuxAChIsVyq4e4kJNmKVHbb255sYzfsPzlqy9dGduu29PsqA7pyvCXPB4NF0gWXyCQrWDeQj0wiO4+KjMefwTXKjSxYjAxbKhwzhieOLVBWPiWACdyoVK9Qgx8sqowMIBzEFz8-x7eGKowgDcQJpbG5oCxNpphFARoIlywYmweBDxmYihdmAQWAGMxqZt1ad2eaAx4W7vERnIGNBhCbBuEATfIBGIechgcFYiR0kD0-XcYTQOwhYO6eV46SgsHgpDR6IxpHEGGgxFEKHaPGIkBQ70g5AgwPh1jh7RQekhRX6BTpegKYTAhDErOEtPJNL0Ek8xgYCjoHAgQ2RkFYnO5-N5emYaEVHiMylF4vAe2lsq54DyoKVnI0cKF6qUmolOsBeu5JtVeU5qvNXit2qltpYcvABMIjqIDLNBmFGrF1s9sDtvrK7jA2AR7ldIvDHr2Ue9+vjidpCecYmTYa1kvTMsz3LzF2z+W4hctqZLQwzPur5jalYKdfdjd15fAHf8eTKnZDFu7NubWcDiO5fpnardDYnZZbc5ZYCC867S8jK6zG7Zg-j-pHdhTxeX0eY2BPg-w9X824vu6v9-nPNPofrz9LrAA9uA5AMuYxBCEgMAlv+4BssIIH1E2lLgCwQA

And something happened when I was setting up scopes for this project, which makes me unable to right mouse clock some components. Exporting as URL and opening in a new tab does not fix this.

The reported error in the javascript console is:

```
Uncaught eo {e: VKb, f: "Exception caught: undefined", g: null, __gwt$backingJsError: Error: Exception caught: undefined
    at vh.sh [as Vd] (circuitjs1-0.js:2446:31)
    at jh (circui…, a: EJb}a: EJb {a: AJb}e: VKb {g: null, __gwt$backingJsError: Error
    at vh.sh [as Vd] (circuitjs1-0.js:2446:87)
    at jh (circuitjs1-0.js:370:26)
    at og (…, $H: 109}f: "Exception caught: undefined"g: null__gwt$backingJsError: Error: Exception caught: undefined
    at vh.sh [as Vd] (circuitjs1-0.js:2446:31)
    at jh (circuitjs1-0.js:370:26)
    at og (circuitjs1-0.js:817:25)
    at eo.rg (circuitjs1-0.js:838:36)
    at eo.xg (circuitjs1-0.js:455:21)
    at eo.ao (circuitjs1-0.js:981:19)
    at new eo (circuitjs1-0.js:296:19)
    at Kn (circuitjs1-0.js:2056:110)
    at Te (circuitjs1-0.js:532:26)
    at Cl (circuitjs1-0.js:2033:145)
    at Ve (circuitjs1-0.js:1796:110)
    at vf.df [as Jd] (circuitjs1-0.js:2445:18251)
    at rG (circuitjs1-0.js:1543:78)
    at HTMLCanvasElement.TH (circuitjs1-0.js:1407:44)
    at Mg (circuitjs1-0.js:845:29)
    at Pg (circuitjs1-0.js:1303:44)
    at HTMLCanvasElement.eval (circuitjs1-0.js:1861:50)__proto__: Object
Kn @ circuitjs1-0.js:2056
Te @ circuitjs1-0.js:532
Cl @ circuitjs1-0.js:2033
Ve @ circuitjs1-0.js:1796
df @ circuitjs1-0.js:2445
rG @ circuitjs1-0.js:1543
TH @ circuitjs1-0.js:1407
Mg @ circuitjs1-0.js:845
Pg @ circuitjs1-0.js:1303
eval @ circuitjs1-0.js:1861
```

This can be reproduced by:
1. Go to https://www.falstad.com/circuit/circuitjs.html?
2. Go to circuits -> passive filters -> band-pass filter
3. Delete the left scope
4. Right click any component
5. _Observe that the right menu is broken now_

A good fix for this would be fixing the root cause, and then applying some logic to automatically fix circuits exported/saved during this broken state

pfalstad, 2021-03-22:
To work around this:

1. File->Export As Text
2. go to the end and delete any lines starting with "o"
3. click Re-Import

pfalstad, 2021-03-23:
Fixed in 2.4.4.  It is kind of weird that we allow "Remove Plot" to remove the only visible plot, but keep the scope.  But I thought it would cause problems to automatically remove a scope when no plots are visible.  For example, you could be in scope properties dialog, and removing the only plot with the intention of adding another, when the scope would disappear on you.

closed 2021-03-23 by pfalstad