sharpie7/circuitjs1 #571: Add language

ppegas, 2021-03-24:
Please add a language - Czech language

[locale_cz.txt](https://github.com/sharpie7/circuitjs1/files/6196920/locale_cz.txt)

Thank you

pfalstad, 2021-03-24:
That was fast.  Take a look here:

http://www.falstad.com/circuit/circuitjs.html?lang=cs

Thanks!  What is your name, so I can put you in the credits?

ppegas, 2021-03-25:
I'm sorry, but I'm writing this text through a compiler.

Thanks for adding. I'm still sending a modified file with multiple expressions. The Czech language selection is missing in the language selection settings on the website.

I will still work on the translation.

Linhart Jiří

PS: In the off-line version, I had the translation as Russian, but after switching the editing of component values ​​was blocked.
It had to be restarted and then everything works fine.
[locale_cz.txt](https://github.com/sharpie7/circuitjs1/files/6202238/locale_cz.txt)

pfalstad, 2021-03-25:
hi,

I've updated the file on my site.  I will fix the language selection settings later.

Jara-da-Cimrman, 2021-03-28:
Hi,
I've been doing a translation for a long time, which I'm constantly updating.
With all due respect, the translation by Mr. Jiří Linhart, which is probably based on a Russian translation, contains incredible nonsense such as:
"Add half an adder" is translated into Czech as "Add half a Viper"
"Add Ring Counter" translated as "Add counter of ringing"
"Ä † uk Converter" = "Ä † uk Converter" !!!!
"A / C source" = "Air Conditioning Source"
"Body Terminal" = "Body Tail"
"Brent-Kung Adder" = "Brent-Kung Viper"
"Caps in Parallel" = "Hat in Parallel"
"carrier" = "Transport Company"
"Combine" = "Harvester"
"Current" = "Actual"
"Flip-Flops" = "Little Frogs"
etc.
.........................

http://www.falstad.com/circuit/circuitjs.html?lang=cs
This is almost fine, but the translation is not complete. There are minor problems, mostly from a stylistic and linguistic point of view.

I do not claim infallibility, but in case of interest I would like to offer an alternative option.

Jara-da-Cimrman, 2021-03-28:
Hi,
I consider the Circuit JS1 to be a great program and that is why I do the translation for a long time and I update it continuously. (just for myself).  But I did not intend to publish it.

With all due respect, the translation by Mr. Jiří Linhart (probably based on a Russian translation) contains incredible nonsenses such as:
"Add half an adder" is translated into Czech as "Add half a Viper"
"Add Ring Counter" translated as "Add counter of ringing"
"Ä † uk Converter" = "Ä † uk Converter" !!!!
"A / C source" = "Air Conditioning Source"
"Body Terminal" = "Body Tail"
"Brent-Kung Adder" = "Brent-Kung Viper"
"Caps in Parallel" = "Hat in Parallel"
"carrier" = "Transport Company"
"Combine" = "Harvester"
"Current" = "Actual"
"Flip-Flops" = "Little Frogs"
etc.
.........................

Unfortunatelly I didn't know that the translation into Czech already exist.
http://www.falstad.com/circuit/circuitjs.html?lang=cs
This is almost fine, but the translation is not complete and there are minor problems, mostly from a stylistic and linguistic point of view.

I don't want to showing off, but in case of interest I could offer my alternative.

Jara-da-Cimrman, 2021-03-28:
Hi,
I consider Circuit JS1 a great program. That's the main reason why I translated it (just for myself) a long time ago and I'm still updating it. But I didn't intend to publish it.

Looking at the translation by Mr. Jiří Linhart (probably based on the Russian translation), I found that (with all due respect) it contains incredible nonsense such as:

"Add half adder" is translated into Czech as "Add half a Viper"
"Add Ring Counter" translated as "Add Counter of Ringing"
"Ä † uk Converter" = "Ä † uk Converter" !!!!
"A / C source" = "Air Conditioning Source"
"Body Terminal" = "Body Tail"
"Brent-Kung Adder" = "Brent-Kung Viper"
"Caps in Parallel" = "Hat in Parallel"
"carrier" = "Transport Company"
"Combine" = "Harvester"
"Current" = "Actual"
"Flip-Flops" = "Little Frogs"
etc.
.........................

At the same time, I also found out that the translation into Czech already exists.
http://www.falstad.com/circuit/circuitjs.html?lang=cs
It is very good, but the translation is not complete and there are some minor problems, mostly from a stylistic and linguistic point of view.
By no means do I want to promote my version, but if you are interested, I can provide it for comparison or compilation.

Thanks.

Karel

Jara-da-Cimrman, 2021-03-28:
Hi,
I consider Circuit JS1 a great program. That's the main reason why I translated it (just for myself) a long time ago and I'm still updating it. But I didn't intend to publish it.

Looking at the translation by Mr. Jiří Linhart (probably based on the Russian translation), I found that (with all due respect) it contains incredible nonsenses such as:

"Add half adder" is translated into Czech as "Add half a Viper"
"Add Ring Counter" translated as "Add Counter of Ringing"
"Ä † uk Converter" = "Ä † uk Converter" !!!!
"A / C source" = "Air Conditioning Source"
"Body Terminal" = "Body Tail"
"Brent-Kung Adder" = "Brent-Kung Viper"
"Caps in Parallel" = "Hat in Parallel"
"carrier" = "Transport Company"
"Combine" = "Harvester"
"Current" = "Actual"
"Flip-Flops" = "Little Frogs"
etc.
.........................

At the same time, I also found out that the translation into Czech already exists.
http://www.falstad.com/circuit/circuitjs.html?lang=cs
It is very good, but the translation is not complete and there are some minor problems, mostly from a stylistic and linguistic point of view.
By no means do I want to promote my version, but if you are interested, I can provide it for comparison or compilation.

Thanks.

Karel

pfalstad, 2021-03-28:
Hi, I'm confused.  Jiří Linhart's version is the one that is on my website.  You're saying it's very good?  That is the second version he posted, the modified file.  It sounds like the first one was from an automated translator?

If you want to send your version, that would be great.

ppegas, 2021-03-28:
Good day,
Because the program made me very nice and I did not find a translation and I was able to easily find a way to translate the lyrics.
I first translated the translation using a goole compiler, because about 1000 phrases can't be translated correctly right away. I was based first on all the translations and in them what I found phrases for translation,
but now I go through the individual translated parts and adjust the translation according to the meaning in the program and adjust the appearance. I'm still working on the translation. I find other translation options. I do not oppose another translation or help with translation.

JL

ppegas, 2021-03-31:
I'm sending another patched file.
I am gradually learning to work with this program while checking the translations. Is there any description of the use of some components, for example in the "Active Building Blocks"?
Probably going to create your own circuits, isn't there an educational film on the Internet?
Thank you
[locale_cz.txt](https://github.com/sharpie7/circuitjs1/files/6235999/locale_cz.txt)

pfalstad, 2021-03-31:
Thanks I updated the file on my site.

Unfortunately there's just the directions.  Some of the active building blocks have links to additional directions in the edit dialog.

pfalstad, 2021-04-01:
Ok I will take a look at this.  In the meantime, I have moved the Czech translation until we agree it's ready.  You can access it here: http://www.falstad.com/circuit/circuitjs.html?lang=csx

pfalstad, 2021-04-01:
I will fix the non-translateable labels like Min Value, etc

pfalstad, 2021-04-01:
Ok I merged Karel's translation with Linhart's, preferring Karel's where there is a conflict, and put the result here:

http://www.falstad.com/circuit/circuitjs60/locale_csmerged.txt

you can try it here:

http://www.falstad.com/circuit/circuitjs.html?lang=csmerged

Here are the duplicates and malformed lines in Karel's file:

ignoring line: "Custom Transformer
ignoring duplicate: "Delay"="Zpoždění"
ignoring duplicate: "Label"="Štítek"
ignoring duplicate: "Min Value"="Minimální Hodnota"
ignoring duplicate: "No file"="Žádný soubor..."
ignoring duplicate: "Resistance"="Odpor"
ignoring duplicate: "Reverse Current Emission Coefficient (NR)"="NR - emisní proudový koeficient v závěrném směru"
ignoring duplicate: "Voltage Doubler"="Zdvojovač napětí (1)"
ignoring duplicate: "Voltage"="Napětí [V]"
ignoring duplicate: "no file loaded"="soubor nezvolen"

Jara-da-Cimrman, 2021-04-01:
HI,
thanks. You are very prompt. I'll check it out.
For the possibility of use, I am attaching the translated help for Custom Transformer, Mosfet (beta), emission coefficient. 

[Custom_transformer_Description_CZ_translation.txt](https://github.com/sharpie7/circuitjs1/files/6241340/Custom_transformer_Description_CZ_translation.txt)

[Diode_LED_Model_Calculator_CZ_translation.txt](https://github.com/sharpie7/circuitjs1/files/6241342/Diode_LED_Model_Calculator_CZ_translation.txt)

[Beta_MOSFET_Description_CZ_translation.txt](https://github.com/sharpie7/circuitjs1/files/6241343/Beta_MOSFET_Description_CZ_translation.txt)

pfalstad, 2021-04-01:
Thanks.  I don't yet have a good way to localize these files..  I'll try to come up with something.

SEVA77, 2021-04-04:
> Looking at the translation by Mr. Jiří Linhart (probably based on the Russian translation), I found that (with all due respect) it contains incredible nonsenses...

@ppegas I don't recommend translating from Russian into Czech language. Some parameters in English have other names in Russian with other meanings of separate words. For example "Parallel Capacitance" and "Series Capacitance" for crystal from English language in Russian this will be "Static Capacitance" and "Dynamic Capacitance" (literal translation from Russian).

The same can be in your language. I recommend finding the correspondences of the components parameters in English in the reference books in your language, paying attention to the meaning of the parameter.

> That is why I have contacted Mr. David Čada, electronic development engineer, who promised to do supervision of the technical terminology.

Very good idea!

pfalstad, 2021-04-24:
Ok you can select Czech from the Options menu now, but it does not load it automatically if you have Czech as the default language in your browser.

closed 2022-07-11 by pfalstad