sharpie7/circuitjs1 #572: AC motor?

K0rrid0r, 2021-03-25:
there is a DC motor but not an AC motor, would also be nice if the glyph could spin.

what do you think?

MarinoHappiness, 2021-04-02:
I think it would be great to have a standard 1-phase and 3-phase AC motor. There are many types of speed controls for AC motors that could be simulated.

pfalstad, 2021-12-11:
The glyph does spin.  It's just really slow.

derestle-htwg, 2023-10-04:
It also would be great to link a quadrature encoder to the motor.
https://github.com/sharpie7/circuitjs1/issues/270
It also would be great to add some specific values to the motor,

Friction Linear,
Friction Quadratic,
Inertia,
Weight (Like an elevator, a constant force in one direction, Motor in one direction, Generator in the other)