sharpie7/circuitjs1 #575: subcircuit with 7 segments led output

goldenshaw, 2021-03-30:
I can build a multi-bit adder using the subcircuit feature, which can hide a lot detail inside the component, it's great. 
But when I try to combine the 7 segment decoder with the 7 segment LED, so to make a simple component which only need four logic input, how can I label the LED output? I can only label the four input node, and the resulting subcircuit component has no LED display as its output

pfalstad, 2021-03-30:
Hi, it sounds like you want to make a subcircuit with an LED display in it, and hope that the LED display will be visible when using the subcircuit?  Unfortunately that's not possible.

goldenshaw, 2021-03-31:
Thanks for your reply, yes, I want it be visible, sorry to hear that's impossible...

closed 2021-03-31 by goldenshaw