sharpie7/circuitjs1 #580: Bug- exported Link shows only white circuit

TimFFischer, 2021-04-12:
I created a small RC-circuit with two labeled nodes and tried to get a correct link to it. 
However, the link shows only a white area but with the correct side bar (checked in Edge and Chrome on WIN10). 

Falstad Code 

$ 1 0.000005 10.20027730826997 50 5 43 5e-11
r 192 112 304 112 0 1000
c 304 112 304 192 0 0.00001 0.001 0.001
g 304 192 304 240 0 0
207 304 112 384 112 5 Out
207 192 112 160 112 5 In
207 48 112 80 112 5 In
v -80 112 0 112 0 0 60.94 5 0 0 0.5
g -80 112 -80 240 0 0
s 0 112 48 112 0 1 false
o 3 64 0 4098 5 0.1 0 2 6 0
38 6 3 1 1000 Frequency

Link:

http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgpABZsKBTAWjDACgAncYlcMX3DT68qYeGwDGIQcOmQhYHlCix4EGHHWr2AczkKlMlDSqm2acjNYC8C-tQDyAVwAu5yOUW9r4QqPtIAJIAdu7kNHiyeP68QaEAbiBM0bIxylR+0MRCSKYqCGx6yWnFIMZ5kGwAzso+EangIABmAIYANlUMbAD20iCEQlQmxJG50Opl-VBs2JGEfRBicCAAYhwMAI5ODMESAJ5sQA

pfalstad, 2021-04-12:
the last line (starting with 38) is the problem.  Remove that, and it works fine.  There is a slider there that controls frequency.  It looks like one of the voltage sources used to be AC, but now it's DC, so there is no way to control frequency.  We need to handle the error gracefully.

http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgpABZsKBTAWjDACgAncYlcMX3DT68qYeGwDGIQcOmQhYHlCix4EGHHWr2AczkKlMlDSqm2acjNYC8C-tQDyAVwAu5yOUW9r4QqPtIAJIAdu7kNHiyeP68QaEAbiBM0bIxylR+0MRCSKYqCGx6yWnFIMZ5kGwAzso+EangIABmAIYANlUMbAD20iCEQlQmxJG50Opl-VBsQA

TimFFischer, 2021-04-14:
You are right - it is now working with an AC source. 
The strange think is, that the code wasn't changed by me. So, somehow circuitjs created the wrong code.
Unluckily, I'm not able to reproduce the bug :-/

pfalstad, 2021-04-14:
You didn't do anything with sliders?  Did you use one of the examples as a starting point?

TimFFischer, 2021-04-18:
I started with the standard RLC-circuit, but deleted everything (so no slider visible). 
In the final state the circuit had an AC source,. At this point I added the slider via the rightclick-menu. 
I try to stay aware and will come back, when something similar happens again.

pfalstad, 2021-04-24:
fixed

closed 2021-04-24 by pfalstad