sharpie7/circuitjs1 #581: Problem creating a new subcircuit based on a previous subcircuit

jhkhouri62, 2021-04-14:
I have created a 4bit Adder subcircuit. Then I tried to create a new subcircuit for a BCD adder using the 4bit Adder. 
No error so far. When inserting a new instance of this BCD adder in a new circuit a Matrix Error message pops-up.

May I have any help?

pfalstad, 2021-04-14:
Hi could you post all the circuits involved?  Use File->Export As Link

jhkhouri62, 2021-04-14:
4 BIT ADDER SUBCIRCUIT: https://tinyurl.com/yztuoccf
BIN TO BCD SUBCIRCUIT:  https://tinyurl.com/yz9f789a

pfalstad, 2021-04-14:
The simulator is confused because the Cin pin is left unconnected.  If you ground that it works:

https://tinyurl.com/yexzrlbh

It's supposed to detect that the pin is unconnected and take care of it.  But for some reason it doesn't detect that when enclosed in a subcircuit.

jhkhouri62, 2021-04-15:
Thank you!! It seems it is solved!! 
It is a great simulator. I am using with my students.

pfalstad, 2021-04-24:
fixed by giving an error when pins are unconnected

closed 2021-04-24 by pfalstad