sharpie7/circuitjs1 #585: Translation fail for sub-items in "Menu-Circuits"

goldenshaw, 2021-04-16:
sub-items in "Menu-Circuits" not translated, although items exists in locale file
[image]

goldenshaw, 2021-04-16:
[locale_zh.txt](https://github.com/sharpie7/circuitjs1/files/6325865/locale_zh.txt)

pfalstad, 2021-04-16:
It's working for me.  Maybe you need to clear your cache?

Thank you for this locale_zh.txt file!  Is it ready to use?  Can I add it to the simulator?

[image]

goldenshaw, 2021-04-18:
try clear cache, but it doesn't work; also try re-import, problem still...
not familiar with gwt, can't feature out why. I notice that the project required J2SE-1.5, I use jdk 1.8 instead, cause I don't have a jdk 1.5, does it matter?

goldenshaw, 2021-04-18:
here is the newest locale zh file, add some newly add item in latest version, such as `New Blank Circuit`, `Toggle Full Screen`, etc.

it is ready to use, can be added to the simulator, thank you.
[locale_zh.txt](https://github.com/sharpie7/circuitjs1/files/6330444/locale_zh.txt)

and here is the code change in `EditOptions.java`, "\u4e2d\u6587" represent "中文(Chinese)"
```git diff
--- a/src/com/lushprojects/circuitjs1/client/EditOptions.java
+++ b/src/com/lushprojects/circuitjs1/client/EditOptions.java
@@ -47,6 +47,7 @@
 	            ei.choice.add("Polski");
 	            ei.choice.add("Português");
 	            ei.choice.add("\u0420\u0443\u0441\u0441\u043a\u0438\u0439"); // Russian 
+	            ei.choice.add("\u4e2d\u6587"); // Chinese 
 	            return ei;
 		}
 		
@@ -90,6 +91,7 @@
 		    	case 8: langString = "pl"; break;
 			case 9: langString = "pt"; break;
 		    	case 10: langString = "ru"; break;
+		    	case 11: langString = "zh"; break;
 		    	}
 		    	if (langString == null)
 		    	    return;
```

goldenshaw, 2021-04-18:
I read the source code, and now I know why, problem exists in `setuplist.txt` and git configuration:
[image]
it seems that when checkout, git automatically change the line separator \n to \r\n, then the `if (b[l+p] == '\n')` statement fail, when change it to `if (b[l+p] == '\n' || b[l+p] == '\r')`, it's OK now.

closed 2021-04-18 by goldenshaw

pfalstad, 2021-04-18:
excellent!  you can see it here:

http://www.falstad.com/circuit/circuitjs.html?lang=zh

I haven't added it to the menu yet; I will do that soon.  Thanks!  What's your name, so I can add you to the credits?

goldenshaw, 2021-04-19:
Thank you. Actually most of the items are not translated by me, I found it here: https://gitee.com/hgcserver/circuitjs1-zh, the author's name is 王逸伦, I fix some misspell, typo and add some newly items.

I also found some other Chinese translated version in other website host in China, they just translate and publish in their own site, I don't know why they not sending the translation to you.

My name is 肖国栋(Xiao Guodong), Xiao is the family name, I don't have a English name, you can call me "Gordon Shaw" which sounds like my Chinese name, or "Golden Shaw" like this github username.