sharpie7/circuitjs1 #588: A 3-phase rectifier circuit outputs entirely incorrect voltage.

roflcopter4, 2021-04-22:
Hi. I can't say I have the slightest clue what the problem could be here (maybe it's me? that's usually a good guess), but I'm at least mostly confident that a bridge rectifier should not output a voltage higher than the peak of it's AC input. So this might be worth reporting just in case. For what it's worth, I spend a couple of hours trying variations of this circuit and got this error every time, so it's not something stupid like putting the ground somewhere odd.

I slapped together a simple full wave 3-phase rectifier, input voltages being 12Vrms 120deg out of phase, earlier (for no particular reason) and was rather surprised to find that the peak of the output DC is around 28V. Assuming I can still do math I do believe 28 is higher than the ~17V peak of a 12Vrms wave. As I said, either this mistake is mine somehow, or something... weird is going on.

Circuit: https://tinyurl.com/yfc2c6ja

Additionally it's curious that (nearly) the same circuit, hastily modified to a half wave rectifier, outputs the expected result.

https://tinyurl.com/ygybmgdt

pfalstad, 2021-04-22:
According to this page: https://www.electronics-tutorials.ws/power/three-phase-rectification.html  A full-wave three phase rectifier should have average output 1.65 * the peak supply voltage, which in this case is about 28V.

That formula doesn't take into account voltage drops across the diodes, so if you subtract out two of those, you get about 26.6 V average, which looks pretty close.

The output voltage at all times is the difference between the highest and lowest input voltages.  So for example when the green waveform hits its peak, the output is the green voltage (about 17V) minus the minimum of the other two voltages (both at about -8.4V) or about 25.4 V.  We're off by about 1 V because of the diode voltage drops.

roflcopter4, 2021-04-22:
It's times like this that I'm grateful for my habit of drawing my sword halfway before doing anything. Much easier to fall on it later when it becomes apparent to all that I'm a moron.

Admittedly, in a fit of distrust and/or existential despair I tested the circuit in spice, which gives the same result (plus some wobbliness from the countless parameters for the diodes, etc).

This circuit sorta makes it obvious that the voltage across the resistor will be higher than any of the peaks of the individual phases. Strong emphasis on "sorta".

https://tinyurl.com/yjfqdgjw

Anyway closing this. Sorry to waste your time. Thank you for so quickly replying too.

closed 2021-04-22 by roflcopter4