sharpie7/circuitjs1 #589: Flyback DC/DC Converter

JonasForssell, 2021-04-24:
Dear Sir,
I am trying to model a flyback DC/DC converter, but I cannot make the transformer operate like a coupled inductor.
Is there a reference model available or can I create a coupled inverter somehow with the inductors available?

Thanks for an excellent program!
/Jonas

pfalstad, 2021-04-24:
Why isn't a transformer working for you?

JonasForssell, 2021-04-25:
Hello Paul,
I may be wrong here, but when I terminate the current flow on the primary side, the energy stored in the transformer is supposed to be released on the secondary side. It feels like energy is not stored in the transformer but feel free to correct me.

https://tinyurl.com/yft9auzt

pfalstad, 2021-04-25:
You need to swap the secondary polarity like so:

https://tinyurl.com/yhejzgea

JonasForssell, 2021-04-27:
Works perfectly now Paul. Thanks!

By the way, is there a way to reduce the oscilloscope resolution per division further? I am max zoomed in according to the slider (5us / div) , but would like to be even closer.

pfalstad, 2021-04-27:
That is strange because for me, the max zoom is 100 us/div.  But in any case, to zoom in closer, you need to reduce the time step size.  Go to Options->Other Options and make the time step smaller.  It can be as small as you want.  It will make the circuit run slower but you will see more detail.

JonasForssell, 2021-04-30:
Thank you for the help Paul,

Here are three versions of converters. I am highly impressed by your program:

Flyback:  https://tinyurl.com/yf4ogwe3
Forward: https://tinyurl.com/yhgdm5el
LLC:  https://tinyurl.com/yztjcz66

My next step would be to try to implement digital control of the LLC.
I have put in a feature request in able to do this, but are open for other ways if they are available.

many thanks
/Jonas

closed 2021-12-11 by pfalstad