sharpie7/circuitjs1 #592: Firefox memory hogging

dzarda, 2021-04-25:
- FF 88 64bit
- Windows 10

It appears we're allocating a lot of objects during the run. Even with the RLC circuit the memory consumption tops at 2.3/8G as seen here:

[image]

The problem manifests itself as repeating periods of normal operation followed by drastic reduction in performance characterizable as periodic denial of service. I was not able to pinpoint *what* the allocations pertain to.

[image]

FF profile here [profile.zip](https://github.com/sharpie7/circuitjs1/files/6371652/profile.zip)

FF memory snapshot here [45260033.zip](https://github.com/sharpie7/circuitjs1/files/6371656/45260033.zip)

Not encoutered in Chrome (different GC policy?)

[image]

pfalstad, 2021-04-25:
Someone reported this already as #490, but that got closed.

I think it is this bug: https://bugzilla.mozilla.org/show_bug.cgi?id=1586495

The more resistors in the circuit, the more memory is leaked.  Only happens on Windows, not Mac.

As a workaround, uncheck "Options->Show Voltage".  I'm not sure if we can work around this in the code.  The CanvasGradient objects doesn't seem to be very reusable, and there's no way to explicitly free them.

dzarda, 2021-04-25:
Okey thanks, I was wondering if this is a FF defect after I saw low memory usage in the mem snapshot.

dzarda, 2021-04-25:
Commented there https://bugzilla.mozilla.org/show_bug.cgi?id=1586495

dzarda, 2021-05-04:
Wow they fixed the problem in latest Beta! Closing

closed 2021-05-04 by dzarda