sharpie7/circuitjs1 #594: Total resistance not shown if two voltage sources exist in on sketch

K0rrid0r, 2021-05-03:
If two voltage sources exist in one sketch total resistance is not shown for any circuit.

[image]
everything like normal here.

[image]
but what's up with this?

also why are these two circuits showing completely different values?
[image]

pfalstad, 2021-05-03:
Yes, we only show resistance if there's only one voltage source in the circuit.  Because otherwise we'd have to analyze the circuit more carefully to determine whether the voltage sources are connected, and this feature didn't seem important enough to make that worth the effort.

These two circuits do not have explicit ground, so the simulator has to choose, and in this case it chose poorly.  Add a ground node to both, and then they will be the same: https://tinyurl.com/yfooogcp

K0rrid0r, 2022-03-19:
has this been taken away in recent updates?

[image]