sharpie7/circuitjs1 #595: Could you make a PNP Optocoupler in the Active Building Blocks?

spaddyconnelly, 2021-05-05:
You have an NPN version. Was wondering if you could make a PNP. This would be helpful for a demonstration for me. 

Thanks!!

pfalstad, 2021-05-06:
I'm having trouble finding any information about PNP optocouplers.  Do you have a particular device in mind, so I could check out the datasheet?

spaddyconnelly, 2021-05-09:
> Hello Paul,
>
> Sorry for the late reply but big thanks for getting back to me! What I was
> thinking of is something we use in the automation industry which
> technically would be called proximity/photo sensors or photointerrupts, but
> the idea is the same. We use them for sensing the physical limits of travel
> for mechanical stages. Here are some screenshots of how most manufacturers
> give the schematics for either NPN or PNP type. I've attached the .pdf's as
> well. However for a good list you can search this link on Digikey:
>
> https://www.digikey.com/en/products/filter/optical-sensors-photointerrupters-slot-type-transistor-output/548
>
> [image: image.png]
>
> Note: some of the newer limit switches have two loads (or can track two
> distances within the switch) but this isn't necessary.
> [image: image.png]
> [image: image.png]
>
> Generally they're 5 or 24V and draw roughly 30-50mA max.
>
> Thanks again for taking the time to reach out to me and I look forward to
> using your tool online!  Really glad I found it. And sometimes wish I had
> found this tool if it existed back when I was in engineering school!
> Shawn Connelly

Hello Paul,

I got the auto reply for attaching too large of datasheets. Here was my
message below without attachments.

 Sorry for the late reply but big thanks for getting back to me! What I was
thinking of is something we use in the automation industry which
technically would be called proximity/photo sensors or photointerrupts, but
the idea is the same. We use them for sensing the physical limits of travel
for mechanical stages. Here are some screenshots of how most manufacturers
give the schematics for either NPN or PNP type. I've attached the .pdf's as
well. However for a good list you can search this link on Digikey:
https://www.digikey.com/en/products/filter/optical-sensors-photointerrupters-slot-type-transistor-output/548

For some specific sensors you can see they sell both NPN/PNP types which
you can see both in the datasheet pictures. Note some of these newer ones
have dual outputs to sense two distances, but that's not necessary.
Here are some that have that particular type of datasheet:
https://www.digikey.com/en/products/detail/panasonic-industrial-automation-sales/PM-K44-C3/3897056
https://www.digikey.com/en/products/detail/omron-automation-and-safety/EE-SX950-W-3M/4986424

Thanks again for reaching out to me and I look forward to using your site!!
Shawn Connelly

ormaaj, 2021-05-14:
CMOS optos also exist [FOD3184](https://www.onsemi.com/products/optoelectronics/igbt-mosfet-gate-drivers-optocouplers/fod3184)

I think all of these can be modelled with simple subcircuits.

ArmoredRodent, 2026-11-15:
You are not really helping your case here. The EE-SX950-W 3M is a npn module. 

[image]

He needs a datasheet. For example. [MCT9001](https://www.onsemi.com/pdf/datasheet/mct9001-d.pdf)

Odds are most of the devices in the list are npn with a npn to pnp transistor on the output. Get us a datasheet or spice simulation of a specific part and I will build you the part. Giving us a module, which can do npn or pnp, does not mean a pnp optocoupler is being used.

[Transistor Output Optocouplers (2,226 ​Products)](https://www.mouser.ca/en/c/optoelectronics/optocouplers-photocouplers/transistor-output-optocouplers/?q=optocoupler%20npn)

And if you check Output Type, they are all npn or npn darlington.