sharpie7/circuitjs1 #598: 741 broken

infinity-computers-dot-net, 2021-05-12:
[image]

....what is there to say?

pfalstad, 2021-05-12:
I'm not seeing that.  When I try it I get 960mV.  When I switch the op-amp model to LM324 v1 I get that, is that what you're using?  But LM324 v2 gives me 17.931mV.  So now you see why there's a v2.  Maybe I should hide the broken v1 model.

infinity-computers-dot-net, 2021-05-12:
Well, this is interesting.

_Now_ when I try it - on a fresh canvas - it only happens with the LM324.

But _before_, it happened on both types. Definitely. Without a shadow of a doubt. I specifically checked.

So most likely, there is a bug in the "changing of the model" code, rather than in the 741 code.

Btw, I don't see a v2 LM324.

pfalstad, 2021-05-12:
Are you using lushprojects.com?  Use http://www.falstad.com/circuit/ , lushprojects is a little behind.

pfalstad, 2021-05-13:
I can't reproduce the "changing of the model" issue.  Maybe you can get it to happen again?  What were you doing just before that?  If it happens again, see if it persists if you reset or if you export/import.

infinity-computers-dot-net, 2021-05-13:
I was trying to design an op-amp-based current sensor (and finding it didn't work properly when I set the lower bound of the op-amp to 1.5v).

I shall try that other site and see how I get on. Thank you.

infinity-computers-dot-net, 2021-05-13:
I have managed to (sort of) [reproduce the problem on the other site.](https://tinyurl.com/yg3e27p9)

Notice that the voltage on the negative exceeds the voltage on the positive, and yet the output is high.

pfalstad, 2021-05-13:
Thanks!  I tried using the 741 model I found on wikipedia and got the same result: https://tinyurl.com/yjqcnbju

I would not be surprised if this is how a real 741 behaves.  The input voltage range is only supposed to be -Vs+2V to +Vs-2V.

The LM324 has problems too.  I would just use the ideal op-amp.

nulldg, 2021-07-10:
> Maybe I should hide the broken v1 model.

I think this is a good idea.

closed 2022-02-19 by pfalstad