sharpie7/circuitjs1 #599: Simulating Power Source

MarinoHappiness, 2021-05-12:
Hello,

How can I simulate a power source with the following specifications?

- Maximum power: 250 W
- Maximum voltage: 30 V
- Maximum current: 8 A

I'm trying to simulate a [solar panel module](https://www.thesolarnerd.com/guide/understanding-solar-panel-specs/). 

Thank you,
Marino

pfalstad, 2021-05-13:
Here is a current-limited DC source with max voltage of 30V and max current of 8 A (which limits power to 240 W).  The resistor is just a test load.

https://tinyurl.com/yg5zxyhj

I'm not sure if this is a realistic model of a solar panel.

MarinoHappiness, 2021-05-13:
Thank you @pfalstad. The complete model has two control input signals--the **irradiance applied to the solar cell** (W/m2) and the **cell temperature** (°C)--as well as many different parameters (cells per module, max voltage, max current, etc.). Check out the [PV Array block](https://www.mathworks.com/help/physmod/sps/powersys/ref/pvarray.html) used in MATLAB.