sharpie7/circuitjs1 #603: Translation submit

dcadc, 2021-05-23:
Hi,
I would like to contribute the translation of Mandarin (Chinese) in Taiwan.
[locale_zhtw.txt](https://github.com/sharpie7/circuitjs1/files/6527645/locale_zhtw.txt) 

It was hard to setup the GWT compiler to verify my translation, so I just translated using [locale_zh.txt](https://github.com/pfalstad/circuitjs1/blob/master/src/com/lushprojects/circuitjs1/public/locale_zh.txt), but found out there are still something not translated on the online demonstration.
I was wondering if there is a full translation template somewhere, thanks!

pfalstad, 2021-05-23:
Thank you!  You can see it here: http://www.falstad.com/circuit/circuitjs.html?lang=zhtw

Here is a list of missing strings.

[missing_zhtw.txt](https://github.com/sharpie7/circuitjs1/files/6527672/missing_zhtw.txt)

dcadc, 2021-05-23:
Newer version with missing strings translated.

[locale_zhtw.txt](https://github.com/sharpie7/circuitjs1/files/6527953/locale_zhtw.txt)

dcadc, 2021-05-23:
Newer version with missing strings translated.

[locale_zhtw.txt](https://github.com/sharpie7/circuitjs1/files/6527961/locale_zhtw.txt)

dcadc, 2021-05-23:
Latest fixed version.

[locale_zhtw.txt](https://github.com/sharpie7/circuitjs1/files/6527996/locale_zhtw.txt)

dcadc, 2021-05-23:
Latest fixed version.

[locale_zhtw.txt](https://github.com/sharpie7/circuitjs1/files/6527998/locale_zhtw.txt)

dcadc, 2021-05-23:
Latest fixed version.

[locale_zhtw.txt](https://github.com/sharpie7/circuitjs1/files/6528527/locale_zhtw.txt)

pfalstad, 2021-05-23:
I have updated my site.  You may have to clear your cache.

What should I use as the language name, for the language chooser in the other options dialog?

dcadc, 2021-05-24:
If we are just adding a new language option, it should be '中文 (台灣)' 

But to be more inclusive (for potential future translations), I suggest we rename '中文' to '中文 (中国大陆)' to be distinct from zh-tw 

That suggestion is based on the de facto translation used on [Wikipedia](https://zh.wikipedia.org/wiki/Wikipedia:%E5%9C%B0%E5%8D%80%E8%A9%9E%E8%99%95%E7%90%86).

pfalstad, 2021-05-24:
This is traditional Chinese, yes?  Would this also be the best choice for Singapore?

pfalstad, 2021-05-24:
Never mind, I see that it wouldn't

pfalstad, 2021-05-24:
The code doesn't support language codes with a hyphen currently..  I will add a special case for Taiwan

pfalstad, 2021-05-24:
Oops that list of missing strings I sent was not up to date.  Here is the up to date list.

"<Strong>RUN</Strong>&nbsp;/&nbsp;Stop"
"Can't load canvas2svg.js."
"Copy to Clipboard"
"Download last played audio"
"File name:"

dcadc, 2021-05-25:
Yes, this is Traditional Chinese, it might look similar to zh-hk (Hong Kong) and zh-mo (Macao), but there are still a lot of differences in terminologies. As far as I know, Singaporean and Malaysian use Simplified Chinese instead of this.

Everything looks great, I think this will be the final version of translation file.
[locale_zhtw.txt](https://github.com/sharpie7/circuitjs1/files/6536761/locale_zhtw.txt)

pfalstad, 2021-05-25:
Thanks, updated

closed 2021-06-22 by pfalstad