sharpie7/circuitjs1 #605: analogWrite for AVR8js version

JonasForssell, 2021-05-31:
Just tried out the AVR8js version. This combo with falstad makes it a very very powerful combination indeed. Thanks for providing this!

A couple of questions and suggestions:

1. analogWrite support
analogWrite does not seem to be supported at the moment. Any plans to make this happen? 
analogWrite is a nice way of generating an adjustable PWM.

2. Saving
How do you recommend I save the model? The shortURLlink approach for falstad is great and if the code could be included there it would be great. Preferrable in point 3 below.

3. Model
This is a demo. Any plans to include the AVR as a simple chip component in falstad? 
The code could be written in the editor in a similar manner that we define input values for other components today. It does not have to be constantly visible. Just when you doubleclick on the component. This would make for a unified interface with the other parts of falstad.

Thanks
/Jonas

pfalstad, 2021-05-31:
analogWrite is supported.  For example, go to http://www.falstad.com/circuit/avr8js/ and change the loop() code to the following:

void loop() {
  analogWrite(9, 64);
  analogWrite(10, 128);
  analogWrite(11, 192);
}

Then run the simulation, and slow the simulation speed down so you can see the LED's blinking.

Yes, I need to figure out some way to integrate the AVR simulator more tightly, so you can at least save the code with the circuit.  Right now you have to manually save your code separately.

nulldg, 2021-06-01:
> Yes, I need to figure out some way to integrate the AVR simulator more tightly

I absolutely love the AVR8js demo and it's been genuinely something I've wanted for a long time, but having two separate systems (circuitjs + wokwi) does prove to be rather cumbersome and limiting. Saving and loading being consequently separate is definitely one of the problems that arise from this.

Having some kind of programmable microcontroller built into circuitjs as a component would be a dream come true, even if the circuitjs component is not based on any real microcontrollers. I can imagine it would be a lot of work to implement, though.

jpk73, 2021-09-09:
Hi, I love your circuti simulator and jave just a short question: how can I open the AVR8js version in stand alone mode?

pfalstad, 2021-09-10:
You can't open the AVR8js version in offline mode, unfortunately.  The AVR8js simulator connects to the wokwi website to do the compilation.  I don't know if that part can even be made offline or if the license allows for it.

jpk73, 2021-09-14:
Hi, it seems it's possible: see https://github.com/wokwi/avr8js

pfalstad, 2021-09-14:
Yes we're using that.  But I don't see a compiler in there.

MarkMegarry, 2021-11-22:
> Hi, I love your circuit simulator and have just a short question: how can I open the AVR8js version offline in stand alone mode?

As defined in /war/avr8js/src.avr8js.js, the buildHex() function sends a JSON string including the user code to https://hexi.wokwi.com/build.  In order to build an offline version of the Arduino simulator, you would have to use a different compiler which works offline.

It might be possible to create a fork which sends the Arduino source code to a local Arduino compiler (if you don't mind finding a way to feed the user-input Arduino code into your command line from the web browser) , formats the resulting hex file as a JSON string and then puts that back into the web app to be simulated with AVR8js.  That would have to replace the code which sends the source code to Wokwi's compiler and records the response.