sharpie7/circuitjs1 #609: Auto Save doesn't work with multiple sessions

dullbananas, 2021-06-03:
I was working on my Bad Apple circuit, and I opened up one of the circuit examples in another window. When I opened the circuit simulator again today, and I used "recover auto-save", it only recovered that circuit example - not the Bad Apple one. Undo didn't work.

I lost a bit of progress of the bad apple circuit! This tragic loss of logic gates is unacceptable! although it's not as bad as [this](https://github.com/MrMEEE/bumblebee-Old-and-abbandoned/issues/123)

If you can't completely fix this, at least show a warning, when opening / modifying a different circuit

pfalstad, 2021-06-04:
You should save your work!  Don't rely on auto-save.  It does give you a warning when closing a window if you haven't saved.

pfalstad, 2021-06-06:
I guess the warning doesn't happen when you close the window on mobile.  (on ipad at least)