sharpie7/circuitjs1 #610: Subcircuit instances can get randomly deleted with Undo

dullbananas, 2021-06-04:


pfalstad, 2021-06-04:
do you have some steps I can follow to reproduce this?

dullbananas, 2021-06-04:
I finally figured out the steps that cause this

1. Connect 3 labeled nodes to an OR gate, and make a subcircuit model from it.
2. Add an instance of that model. Connect one input to a voltage source. Connect the output to a labeled node.
3. Stop the simulation.
4. Duplicate the subcircuit instance, and the labeled node, but not the voltage source. Do not change the labeled node's name.
5. Duplicate again. This time, it refuses to duplicate the subcircuit instance.
6. Add a wire somewhere.
7. Undo. Subcircuit instance disappears.

pfalstad, 2021-06-04:
for (4) you're selecting the subcircuit instance and the labeled node, and duplicating them both together?

what do you mean it "refuses" to duplicate the subcircuit instance..  what does it do instead?  For me step 5 works fine, and undoing in step 7 works as expected.

dullbananas, 2021-06-04:
In step 5, it only duplicates the labeled node. Maybe you need to duplicate more times. Did you do step 3, and add the voltage source in step 2?

pfalstad, 2021-06-04:
Oh I see.  I did do step 3 but I must have accidentally started it again.

There's an exception in the console when you do this.  I've seen that bug once before but didn't realize it was this easy to reproduce.

pfalstad, 2021-06-22:
fixed

closed 2021-06-22 by pfalstad