sharpie7/circuitjs1 #611: labeled nodes

dullbananas, 2021-06-04:
These are centered aligned. They would look better left-aligned, and rotated when vertical
[image]

pfalstad, 2021-06-04:
not sure about the rotation when vertical.  just space them farther apart?

dullbananas, 2021-06-04:
You are right. Maybe it should only be left-aligned when horizontal

dullbananas, 2021-06-04:
or maybe wrap the text when vertical

pfalstad, 2021-06-04:
The idea behind labeled nodes is to identify pins in subcircuits, and to connect parts of a circuit without having to route wires.  So in both cases, the labels should be relatively short.  If you want to add text to describe parts of a circuit, you can create text elements, which can be wrapped onto multiple lines (by using \n to add line breaks)

pfalstad, 2021-12-10:
This is partly fixed but I'm leaving it open because it might be useful to have the node name printed vertically as an option.