sharpie7/circuitjs1 #618: Request for relay symbol update according to IEC-60617

tito125, 2021-06-11:
Hello,

I have made a circuit with relays and time relays, only it becomes unreadable with this type of symbol.
Would it be possible to modify the relay and timer symbols to the international standard IEC-60617.

Relay with NO and NC contact

[image]

Symbol link :
[https://knowledge.autodesk.com/fr/support/autocad-electrical/learn-explore/caas/CloudHelp/cloudhelp/2019/FRA/AutoCAD-Electrical/files/GUID-DA927086-EEAF-48CB-B9B1-634A4B48C8C7-htm.html](url)

Many thanks in advance

nulldg, 2021-06-11:
Unless I'm not being very imaginative, the relevant symbols prescribed by IEC-60617 doesn't appear to be very good for illustrating the internal state of the relay, which is important for the circuit simulation.
We can mitigate the ambiguity by adding some kind of rectangle around the relay. What do you think?
[image]

pfalstad, 2021-06-11:
The IEC symbols are all two- or three-terminal devices, also.  As far as I can tell, the time delay relay symbols don't show the power connections, and the relay symbols don't show the coil.

tito125, 2021-06-28:
Hello,
Yes, indeed, it looks like that for simple relays, I modified the image and put an example of a typical relay that would correspond more to the standard.

Cordially
[image]

nulldg, 2021-06-28:
We've already created and added the change (https://github.com/pfalstad/circuitjs1/pull/13). It's now on paul's public build of circuitjs if you'd like to play with it.
We could probably further improve it by:
- changing the outline to be a dotted rather than a solid.
- an option to move one of the relay coil's terminals to the opposite end of the relay rather than looping back

But this might be getting rather excessive. I think the way it looks now is quite nice.
[image]
[image]

pfalstad, 2021-06-28:
Well we should try to look like the standard symbol as much as possible.  The relays need work anyway so these would be good changes to make.

nulldg, 2021-06-28:
I think we should replace the "Swap Coil Direction" checkbox with a listbox for the different coil styles: default, flipped, and straight. We might need to replace the `FLAG_SWAP_COIL` flag with a style integer to accept more than 2 styles. Or maybe not. I'm not entirely sure how to write this cleanly without changing the `FLAG_SHOW_BOX`'s value we added very recently

pfalstad, 2021-06-28:
yes we should have a listbox (though I would make straight the default).  Internally though, I'd just keep FLAG_SWAP_COIL and add another flag for straight.

nulldg, 2021-06-28:
What happens if both flags are high? What if we want to add more styles?

pfalstad, 2021-06-28:
well the UI code will make sure only valid combinations of flags are allowed.

there's plenty of room for more flags.  how many coil styles can there be?  we're already stuck with FLAG_SWAP_COIL so sticking with flags seems the simplest.

nulldg, 2021-06-28:
Fair enough, I agree. I'm just thinking ahead, because I doubt this was anticipated when the "swap coil direction" option was added. Once we make a change to the dump format, we can't undo it without adding legacy code, but if we ever do need to add another in the future, we could just treat the flags as the bits of a 2-bit integer.

K0rrid0r, 2021-07-16:
This was a great update, thanks!

closed 2021-07-17 by pfalstad