sharpie7/circuitjs1 #619: Grounded transmission line spooky action at a distance

ormaaj, 2021-06-16:
Requiring the ends of the shield to be connected through a worm hole isn't so great (unless the line is actually a closed loop). I wanted to simulate a proper locality-respecting t-line with neither end connected to the circuit ground. I think this does a decent job by using two of them, with a few workarounds:

 * [functiongen.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/6660652/functiongen.circuitjs.txt)

But I think you'd get close to the same by just removing the grounded shield requirement. The shield nodes need to be free to simulate things like transmission line transformers, active shield drivers and other parasitic effects.

pfalstad, 2021-06-22:
Ok I have no idea what's going on this circuit but basically what you're asking for is a transmission line which doesn't have to be grounded, right?

ormaaj, 2021-06-23:
> Ok I have no idea what's going on this circuit but basically what you're asking for is a transmission line which doesn't have to be grounded, right?

Right. Half of that circuit is unrelated (mostly). The left hand side shows a sort-of work around  by using a 2nd line as as a simple analog delay that's terminated to damp reflections from propagating forever in a loop, I copy voltage in and current out, summing the termination current to cancel its influence and isolate the shield output from ground with all the currents accounted for.  I'm not completely certain that accounts for everything though. Distributed capacitance and mutual inductance between the two lines aren't modelled of course so the 4 pin's currents and voltages are only be determined by the impedances of whatever they're connected to.

Your model looks pretty similar to the spice lossless transmission line with the grounded shield behaviour being an obvious difference.

pfalstad, 2021-06-23:
The grounded shield behavior is an attempt to hide some flaws in the model.  It seems the spice model has these same flaws, unless I'm missing something.

Take the following circuit: https://tinyurl.com/yeda6mde

In my test version which has the ground requirement commented out, nothing happens; there's no current anywhere.

I tried it in spice and got the same result:
[tl0.txt](https://github.com/sharpie7/circuitjs1/files/6704620/tl0.txt)

endolith, 2021-11-21:
> The shield nodes need to be free to simulate things like transmission line transformers, active shield drivers and other parasitic effects.

also https://www.youtube.com/watch?v=bHIhgxav9LY :)

ormaaj, 2021-11-21:
> > The shield nodes need to be free to simulate things like transmission line transformers, active shield drivers and other parasitic effects.
> 
> also https://www.youtube.com/watch?v=bHIhgxav9LY :)

Yep. Even if RGLC parameters were given we'd need a 2-d model with a conductor spacing parameter. I've not seen that in a spice-like simulation. A lumped-element approximation with `.k` linked inductors probably wouldn't be suitable either. Likely have to step up to a proper field simulator for that.

pfalstad, 2021-11-22:
I'm not very happy with the end of that video.  I mean, the light bulb would turn on, sure.  But you don't even have to turn on the switch.  You don't even need the battery; any random electromagnetic noise would cause current to flow and the light bulb to light up.

endolith, 2021-11-22:
@pfalstad Yeah it would be a very small current and wouldn't light the bulb visibly regardless, as many comments have pointed out, but this is off topic. 😁

pfalstad, 2021-11-22:
Back on topic, I realized I hadn't tried the spice lossy transmission line model.  I tried 2 of the models in my copy of MacSpice with my test circuit above and they both gave me strange results that don't seem correct.  The other model is just a LC ladder, which I'd rather not use because it would be a lot slower.  If you want to use that, you could build your own.