sharpie7/circuitjs1 #620: Feature Request: Chokes

haydenth, 2021-06-18:
Hello all! 

I'm trying to figure out how to model a choke in circuitjs. Here's an example circuit from wikipedia.
[image]

I don't see this as an option and perhaps its related to using a transformer like in #392 

If this exists and I'm an idiot, please just close this and tell me off :)

pfalstad, 2021-06-18:
An RF choke could be modeled as a small inductor, like 2 millihenries.

The choke in your picture is a common-mode choke, and can be implemented using a transformer with a small inductance.  According to some datasheets I found, the inductance could range from 7 microhenries to 60 millihenries.

haydenth, 2021-06-18:
Perfect, thank you. Didn't realize it was quite that straightforward :)

haydenth, 2021-06-18:
Shortly after I closed this, I found a good document on chokes + equivalent circuits if anyone in the future ever ends up here; 

https://www.murata.com/~/media/webrenewal/products/emc/emifil/knowhow/26to30.ashx

closed 2021-06-18 by haydenth