sharpie7/circuitjs1 #621: Mods for server-less operation?

erantapaa, 2021-06-23:
I am interested in running circuitjs without a web server -- i.e. the files are served directly from the file system.

The generated files in the war directory seem to work (given my limited testing) and I only ran into one small issue -- upon startup you get an error dialog saying that the circuits couldn't be loaded.

It's not a big deal since you can load them using the File -> Open File menu item. But I was just wondering if there are any other server interactions that I might get tripped up on.

(btw - I fixed the can't find circuits error message by just having getSetupList return immediately)

pfalstad, 2021-06-23:
Did you see the offline version here: http://www.falstad.com/circuit/offline/

I think the only other issues you might see are that Export as Link may not generate useful links, and the locale files won't be loaded.

erantapaa, 2021-06-23:
Hi - thanks, I should have mentioned that I want to run it on my Chromebook!

erantapaa, 2021-06-23:
Ok, here's one issue I'm running into...

When trying to run jsinterface.html, it doesn't look like the main page can interact with the iframe. This callback doesn't seem to be executing:

```
iframe.contentWindow.oncircuitjsloaded = simLoaded;
```
and I'm getting CORS errors when trying to access into the iframe:

```
sim = iframe.contentWindow.CircuitJS1;
```
Might have to run a web server after all...

erantapaa, 2021-06-23:
Here's a new idea... is there javascript code I can add to circuitjs.html to access the CircuitJS1 object?

pfalstad, 2021-06-23:
Right, I don't think jsinterface.html is going to work.  I'll look into making the simulator installable on a chromebook, which should solve these problems.

You should be able to access the CircuitJS1 object directly from circuitjs.html.

erantapaa, 2021-06-23:
That would be great!

After playing around with it I figured out you can access CircuitJS1 from top-level JS code. GWT seems to blow away any hard-coded HTML elements you have on the page but I think I work around that.