sharpie7/circuitjs1 #622: Monostable does not work after reset

JonasForssell, 2021-07-08:
I have a small issue with the monostable chip.
It works well until I have had a convergence fail in the model. This happens sometimes when I run outside normal operation.
When this happens I usually reset the simulation but for some reason the Monostable component stops working after this has happened. It is a bit like that it does not seem to be completely reset as it should.

Perhaps a bug?

Many thanks for a fantastic program.

pfalstad, 2021-07-08:
It's working for me, except that when I click reset, both Q and /Q are low, which is wrong.  Is that what you're referring to?  But if I trigger it, it works again.

pfalstad, 2021-07-09:
Oh I see..  If you reset it when Q is high, it will never trigger again.

pfalstad, 2021-07-18:
fixed

closed 2021-07-18 by pfalstad