sharpie7/circuitjs1 #623: Params in Subcircuit

Fvi589, 2021-07-08:
Hi,

Would it be possible to be able to modify the values ​​of the passive elements contained in a subcircuit ?

Best regard
Franck

pfalstad, 2021-07-08:
Hmm, maybe the sliders could be converted to parameters when you create the subcircuit?

Fvi589, 2021-07-09:
yes it would be great, if by sliders, we could change the params

Fvi589, 2021-07-10:
If we put switches in the subcircuit, could we activate them by a slider or other, in the right bar? Example: circuit breaker ON / OFF

Best regard

pfalstad, 2021-07-10:
I was thinking that sliders would just mark the parameters.  If you use the subcircuit, there wouldn't actually be sliders.  You would change the parameters using the component's edit menu.  Of course if you wanted sliders, you could add some using the Sliders... dialog, but normally you wouldn't.

I also wasn't thinking that there would be any on/off choices.  But actually, that is another idea.  Maybe we could use switches instead (or in addition) to mark the parameters.  Any switch in the subcircuit, can be turned on and off using the edit menu.  Not with a slider in the right bar, though.

nulldg, 2021-07-10:
I like this idea, but I wouldn't limit it to passive elements. Plus, the behavior should be documented, otherwise very few will know this feature exists, let alone will know how to use it effectively.

Fvi589, 2021-07-11:
I made a very simple example just to illustrate. My idea is from the subcircuit, which I will use in another assembly, to be able to change the value of R by slider, but also to be able to have access to the switch through the subcircuit, without having to edit the subcircuit.
I do not know if I'm clear, do not hesitate to tell me.

$ 1 0.000005 10.20027730826997 50 5 43 5e-11
r 128 160 240 160 0 1000
s 240 160 368 160 0 1 false
207 368 160 464 160 4 B2
207 128 160 48 160 4 B1
. Essai 0 2 1 2 B1 1 0 2 B2 3 0 3 ResistorElm\s1\s2\rSwitchElm\s2\s3 0\\s1000\s0\\s1\\sfalse
410 192 272 320 272 1 Essai 0\s1000 0\s1\sfalse

ormaaj, 2021-09-13:
Heh, another silly idea might be to abuse subcircuits as anonymous "lambda" functions and let the pin / argument names handle all of the above. That might be a little annoying.

Soon there will be subcircuit y-combinators. *shudder*

pfalstad, 2021-09-13:
People want the slider features to be improved as well (see #504).  If the slider code isn't good enough we could make it better.

Right now we can take any parameter and link it directly to a slider.  We could instead have (1) a list of sliders with labels, variable names, and min/max values; and (2) allow any parameter to be replaced by an expression that references the sliders (using their variable names).  (2) would be done using a UI similar to what we have now, for sliders.

The same could be done for subcircuits.

It's a separate issue, but it also might be a good idea to be able to create voltage/current sensor elements which could be a linked to a variable that could be referenced in any [VC]C[VC]S expression.  People complain about the [VC]C[VC]S's as well, because they don't look anything like standard symbols used for controlled sources, and that would fix that problem.

ormaaj, 2021-09-13:
True making the sliders programmable might well accomplish the same, I like that idea. Also easy since they already have the necessary hooks into the right places.

Also obvious feature - a way to toggle continuous evaluation and "eval once" fixed constants.