sharpie7/circuitjs1 #624: An option to rotate ChipElm by 90 degrees

nulldg, 2021-07-10:
Right now the chips can be flipped in both X and/or Y axes, but if they could be turned by 90 degrees, then theoretically the component can be rotated to *any* cardinal orientation without redundancy.

With just this, each angle can be achieved through the following transformations:
Angle | Transforms
-- | --
0° | 
90° | Rotate
180° | Flip X + Flip Y
270° | Rotate + Flip X + Flip Y

And then to get the flipped variations of each angle, just toggle Flip X and/or Flip Y respectively.
[image]

---
I think this would be trivial to implement because a similar thing has been done for flipping in `ChipElm.Pin.setPoint` (in `ChipElm.java`).
When a ChipElm is marked for 90 degree rotation (via a flag): Immediately after `setupPins` is executed and before the posts are setup and the component is stamped, the chip's width and height would be swapped. Code for rotation would then be written to `ChipElm.Pin.setPoint`, prior to the flipping code.

It'll be important to review all the references to `sizeX` and `sizeY` to make sure the size doesn't get used to calculate/setup *during* the `setupPins` call of any components (rather than afterwards, when it'll be safe to do so), especially for the LED array, whose function is closely tied to its size.

pfalstad, 2021-12-10:
added

closed 2021-12-10 by pfalstad

nulldg, 2021-12-10:
That's one monster of a commit. Excellent work, Paul!