sharpie7/circuitjs1 #626: Export to Spice Netlist?

marcdraco, 2021-07-15:
Hey Iain,

wonder if it's possible to do some sort of netlist export so we can import a *working* schematic into a PCB designer - perhaps KiCAD or similar? Not even sure if this is practical but it should would help!

ormaaj, 2021-07-17:
A spice netlist doesn't help. Netlists discard the schematic data. The circuitjs format is already netlist-like but augmented with the schematic coordinates and optionally the simulation state. Netlist / schematic capture conversion tools are very easy to write. I would not put them into the simulator.

ormaaj, 2021-07-17:
I would however clean up the file by not doing the "number suffix" to float expansion prior to saving the output. That would make tools hard to write, and the file output way larger.

As an aside - looks like @pfalstad put in a stub function to add text compression to the output generator. But then I just shrunk a huge output file to fit into the tinyurl thing by more than you could with compression by just discarding the simulation state, fixing all the `1.00000000000000001` crap, and manually re-importing the text. Probably some other things that could be done to reduce repetition and make the save file a bit saner.

marcdraco, 2021-07-17:
Thanks!

I guess someone with more time on their hands than I have might want to take it up. :)

I'm too old to do much programming (or electronic design) now which is where simulators come in - problem is finding one that's FOSS and reasonably reliable. There's always a (good) chance that I'll flub something when copying things to the PCB layout (Fritzing, KiCAD, etc.) with fairly predictable results. I didn't matter when I could produce my own boards, but the 2-3 week turnaround means that I'm not really willing to leave anything to chance.

Unless I missed something here? I come from the days of valves and stone tablets where we painted etch resist on by hand and then hoped it didn't wash off... you get the idea. CAD makes things accessible to old farts like me that I couldn't even imagine back then. I mean, I can even drop a QNL (?) SMT MCU on a drawing, upload it to China and have it here (assembled) in under a week ready to program and ship! Fortunately my stuff is open source or I'd be using someone else... (cough, cough).

ormaaj, 2021-07-17:
This is a somewhat simplistic simulator. It's not meant to integrate with CAD workflow. The real problem is eeschema is crap, and spice integration isn't really there. That's where the work needs to be done.

ipatrol, 2021-09-09:
It would be helpful to have a short description of the file format already used for import/export. The rest is (theoretically) just a matter of textual transforms.