sharpie7/circuitjs1 #627: Bug- Analog Reciprocal circuit missing

carpetbugs, 2021-07-17:
Hi,

I have noticed that the Analog Reciprocal example circuit is not present in the latest version of the simulator on http://www.falstad.com/circuit/ but it does seem to currently exist on http://lushprojects.com/circuitjs/ .

pfalstad, 2021-07-17:
Apparently it's not a realistic circuit; it doesn't work in spice.  So I took it out.

ormaaj, 2021-07-17:
I think that was based on my comment. The analog reciprocal example was dependant on realistic small-signal behaviour so I tested using extremely low betas, as the default "20m" is still an extremely high value. What I didn't realize at the time is that the "simulate body diode" option is really dumb and (I think) it basically just inserts the "default diode" model in parallel with the mosfet without scaling. The default diode has an extremely unrealistic reverse leakage which dominates at low beta making my test invalidish.

So "not a realistic circuit"... meh kind of a matter of perspective. The mosfet model is unrealistic, but if you want realism then you absolutely must have realistic capacitances too, which requires dealing with parallel capacitors, another issue.

pfalstad, 2021-07-17:
I think the bigger problem with the circuit is that I suspect someone came up with it on their own when playing with the simulator.  I couldn't find it anywhere else.  I want the example circuits to be based on real circuits people use.

20m is high?  I had someone else complain that it was too low!  We decided that 20m was a good value for signal MOSFETs, but for power MOSFETs, it should be 80 (no m).

I should really replace the "simulate body diode" thing using a model from spice.  But all the mosfets in the analog reciprocal circuit have that option turned off, and it wouldn't apply even if it were on, so I'm not sure how that's relevant.

closed 2021-07-18 by carpetbugs