sharpie7/circuitjs1 #632: offline version minor but essential updates

q12a, 2021-07-31:
Hello mister Paul,
First of all, your software is a piece of jewel. I understand you concentrated to make it absolutely an online browser application. 
I'm not saying to turn away from your path, but by adding some little details here and there to the offline version, we the (offline category) users, will be waaaaay more happy. Don't you want us happy?
 My little request for the offline version:
 -A- Please add  a shortcut key for Save As (Shift + S) or/and Save (Ctrl + S)
 -B- Please add  a [-Most Recent Saved Files-] in the File menu. For more easy file handling.
Thank you very much !

pfalstad, 2021-07-31:
There's already a shortcut for Save.  I posted that version on July 11.  You're not seeing it?

q12a, 2021-08-01:
Yes, I just look into it and it is there, Very nice ! 
And thank you. 
A little suggestion though. For an unsaved file, change the Name of the Form as [*Circuit JS1] and after the save is done [Circuit JS1] without the asterisk. Like in the Notepad, it is a good way of announcing you, hey, it is saved or hey, its not saved. Very efficient visual marker. 
If im asking too much, hey, i'm glad you've added this little implementation as it is.
Again, thank you.

pfalstad, 2021-08-01:
Good idea, not sure why I didn't do that.  Maybe there was a problem with changing the title, but I'll look again.

For now, you can look in the lower right corner.  There's a little "saved" indication.

q12a, 2021-08-02:
Yes, now that you told me where to look, i've seen it. To be honest, it is almost invisible there in that always updating and moving text field. That's why I so easily ignore it. It's good that you implement it at least. 
My suggestion, is to make this Saved Attention, very evident. to attract the attention a little bit better. 
You can also make a "status" bar in the bottom of the program and in there specify all the activity, including "file is saved" or "file is changed", etc. 
I will give you the "Word" example again, it is having that status bar on the bottom but they "go to town" with that little cute progress bar. Again, to grab the attention that something is happening. That progress bar is for larger file sizes. For small filesize is almost a blip. But the idea is very good. 
[image]
I am giving you suggestions here, you do whatever you feel is good and fast and productive. And whatever you choose, doesn't matter how it looks, it should do it's job, but a bit more clearly and evident.
 Announce me when is done.
 Im more interested on the "most recent files" but 1 step at a time. Haha.

q12a, 2021-08-02:
1- the asterisk on the title  
2- status bar on the bottom of the program
I'll implement 1 of them or both of them. So, 2 options.  -- both of them = more professional looking ;) --
Go with the asterisk for the moment and see if its doable. It should be easy but you know better what you have there.

q12a, 2021-08-15:
It just pop out to me this little cute and simple implementation,and I share it with you, to influence you further:
[image]
[image]

q12a, 2022-01-03:
Hello again mister Paul Falstad, and happy new year 2022.
(I am using the PC version of your program, not the online version)
For some time I have this problem:
[image]
When I have the component created already in the canvas, I can NOT turn it around at all. Yes I see the little square but it does not rotate. 
The component does rotate only when is First instantiated/Created from its main menu. Thats it. Once it is on the canvas and used, it does not rotate at all. 
 Please, can you check every component in your database, including the NPN and PNP transistors and repair the rotation for me? Some components have the rotation working already, like the diode,LED, Potentiometer,etc. But you forgot about the transistors. 
 Please tell me if you are making it. 
Thank you and continue to do a great work !