sharpie7/circuitjs1 #634: Suggestions regarding capacitors:

Jeppe53, 2021-08-07:
When two or more capacitors are connected in parallel the simulation stops (in real life is it not a problem :) ).
Couldn't it be smart if a capacitor basically had a small resistor (like 1 microohm) in series by default. 
It should be made possible to change the value during editing of the capacitor.

/Jeppe

pfalstad, 2021-08-07:
We could allow this now.  This used to be necessary, but I changed the capacitor model so it's possible now.  We'd just have to do some extra work in the circuit validation to ensure that two capacitors connected in parallel have the same voltage, or else we'd get oscillation.

A resistor in series would work but it would make capacitors twice as expensive to create (an extra circuit node for each capacitor), which would make large circuits run a lot slower.  Probably not a good idea to do that just to handle this unusual case.

nulldg, 2021-08-07:
The capacitors found in simulations are theoretical capacitors, so they exclusively simulate theoretical capacitor behavior and nothing else.
If you need internal resistance, I would recommend simply include a tiny series resistor to each relevant capacitor to explicitly simulate it. We create equivalent circuits like this whenever we need to explicitly simulate something, such as electrical noise, trace inductance, heat, internal resistance, trace resistance, etc.

Jeppe53, 2021-08-08:
Thanks for the responses.
I been using "falstad/circuit" for years and are fully aware how to overcome this minor problem by putting a resistor in series with the "second" capacitor and my students are learning it quickly.
But I also notice that the simulator keep getting improved. Hence my suggestion what could be changed to make the use more "userfriendly".

But Im aware that this will interfere with the principles of using ideal components-models.

ZVSphiler, 2022-01-09:
Maybe allow specifying components as parasitic components to explicit them from drawing. Or integrate them directly into (standard) components just like how it was done in ltspice. Just an idea

Killeroid17, 2023-02-11:
> 

AGREED, because in REAL LIFE every conducotr / wire have at least 0.001 m OHM !

Killeroid17, 2023-02-11:
> 

looks like this problem is an old one and still didn't get a REAL-LIFE- update...

Killeroid17, 2023-02-11:
> We could allow this now. This used to be necessary, but I changed the capacitor model so it's possible now. We'd just have to do some extra work in the circuit validation to ensure that two [...]

or at least eliminate this error and leave it as is, without the simulation being stopped...

esaruoho, 2026-03-03:
Fixed in v3-dev — capacitor loop error removed.