sharpie7/circuitjs1 #637: RC circuit, diode in parallel with resistor

NickNerd, 2021-08-17:
Hi, I was simulating an RC circuit with a diode in parallel with the resistor, and saw some strange behaviour in the simulator.

When the capacitor is discharging in the simulator, it drops to a negative voltage. From my knowledge, I would have expected the capacitor to simply instantaneously discharge to 0 V and not discharge negatively.

Link to circuit: https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWEAOaAmZYCcB2BzkcBmLUhJTEJAFiKoFMBaMMAKACdwA2ScDEDMj5DeYeJFYBjEETSj+s3nJHhY49erBMcIRjEjZMxSCTQJqCMFwj6DrAO4CC3Jc7D8JjwcJlyfEgBMnIStRHhcBcAA5ajBqZAdfeSEiZGp-VgAlJIFIFL80al5ikCKocuLoBETQn1razwj3EP5mqCy6-iwuf0iytCqBQcrqgHsqEC503iLSCdsTLEguM3K0CaJWIA

I made this circuit up on a breadboard and saw the capacitor discharge almost instantaneously to 0 V on the oscilloscope: 
[image]

Is this a bug, or am I missing something?

Thanks,
Nicholas

pfalstad, 2021-08-17:
Switch off trapezoidal approximation...   Right click the capacitor, and uncheck the trapezoidal box.

Trapezoidal will give you better accuracy for something like an LRC circuit or a filter, but in cases like this with sudden large currents, it can create overshoot or instability (oscillation).

NickNerd, 2021-08-20:
Great, unchecking trapezoidal approximation worked for me, thank you very much.

closed 2021-08-20 by NickNerd